esp8266/Arduino #638: IPv6 Support

owendelong, 2015-07-31:
Currently the entire ESP8266 implementation is IPv4 monastack. It would be nice to be able to choose (through compile-time directives or other mechanism) IPv4, IPv6, or IPv4v6 dual stack.

The lack of IPv6 support is already proving a hinderance in some environments where I work and I can only imagine that this will get worse as we go forward. The ARIN free pool is down to fewer than 256 /24 prefixes. Less than 65,000 total available IPv4 addresses for all of north america.

far5893, 2015-07-31:
I suppose that it depend from expressif OS and not from arduino wrapper.

owendelong, 2015-08-20:
Perhaps, but I'm betting if the Arduino community as a whole started pushing for this, Espressif would be more likely to listen than to just me. I've also put this forward with them as well.

igrr, 2015-08-20:
For the record, they have already released lwIP source code for 1.3.0 SDK.

mst78, 2015-12-15:
There seems to be IPV6 support in the espressif RTOS SDK1.3:
http://bbs.espressif.com/viewtopic.php?t=619#p4444

@Igrr: Is it possible to integrate IPV6 now? Do you have any plans of doing it?

ccoenen, 2016-01-07:
I'm interested in IPv6 support, as well!

Toshik, 2016-02-19:
Any news on IPv6?

Skeen, 2016-03-29:
+1 for IPv6

igrr, 2016-03-29:
IPv6 support isn't present in the non-OS SDK, which is used by this project.
RTOS SDK, on the other hand, has IPv6. An Arduino core for the ESP8266/ESP32 based on RTOS SDK will be released as a separate project.

mst78, 2016-05-07:
Igrr, do you have any idea when the first version of Arduino core for the ESP8266/ESP32 based on RTOS SDK will be available?
Have you already started working on it?

igrr, 2016-05-07:
Yes, i was in fact working on things related to IPv6 support in ESP32 Arduino core yesterday :) 
I can't give you any concrete date unfortunately. It's going to be available within next two months, this is somewhat gated by the release of new SDK version.

For now, there is a publicly available ESP32 core by @me-no-dev. It has partial IPv6 support. LwIP stack has support for IPv6, but it is not complete (missing IPv6 support in DNS client, for instance). On the Arduino side of things, IPAddress class in this core also doesn't support IPv6 addresses.

MyIgel, 2016-09-29:
@igrr Any updates?

igrr, 2016-09-29:
Regarding ipv6 for the 8266 non-OS SDK, I don't have any updates.

FlorianHeigl, 2016-11-28:
ping, ipv4 is deprecated!
I was just checking this to install a few dozen at a conference. it's unlikely they still allow ipv4.
is this something you do at work? can you tell your boss this actually matters?
If it's just out of your free time and energy, then don't worry! :)

kohlsalem, 2016-12-21:
+1 and quite surprised to see a product so recent w/o ipv6

itavero, 2017-01-06:
Has anyone looked at integrating picoTCP?
I'm pretty sure they have had IPv6 support for a while now.
https://github.com/tass-belgium/picotcp

MatthiasLohr, 2017-06-20:
Anything new about this topic? I' interested in IPv6, too...

NiKiZe, 2017-07-29:
Sorry to bump this but, any update?

ttytyper, 2017-07-29:
In case nobody has noticed, there's a $30 bounty on this issue: https://www.bountysource.com/issues/25224540-ipv6-support/backers

FlorianHeigl, 2017-07-29:
Do you think the bounty will help? just wondering since there's no reply here ever since...
Would not mind throwing more at it. I bet most of us would do that.

krmax44, 2017-12-29:
Has there been any progress since then? IPv6 is a very important thing in the IOT world.

d-a-v, 2017-12-30:
Doable. At least tryable. Buy me time, here are some [hints](https://github.com/d-a-v/esp82xx-nonos-linklayer/issues/8#issuecomment-354570604). core-2.4.0 has first to happen though ;)

devyte, 2017-12-30:
The precursor was migrating to lwip2, and that is in. Now it just needs to be stabilized. After that, the glue layer between the core and lwip needs some changes. After that, our socket layer needs to get support. After that, the rest of the core and libs.
It will take a (long) while, but now at least it's possible.

FlorianHeigl, 2017-12-31:
Thank you @d-a-v and @devyte for giving outsiders some insight and also for having it on your mind, and for all that you did towards it. seems a lot of pieces need to be moved :)

d-a-v, 2018-09-14:
#5136

C0rn3j, 2018-09-14:
Just decided to check up on this for the first time and I see a PR 2 hours ago... great job!

C0rn3j, 2018-09-19:
@d-a-v would love to test but failing to find the "Arduino application directory" on Arch Linux (Arduino 1.8.5). (Trying to follow this https://github.com/esp8266/Arduino#using-git-version )

Seems like the esp8266 package, when installed through the IDE, installs under ~/.arduino15, but libraries like IRremoteESP8266 install under ~/Arduino.

I've tried installing it under the already-installed esp8266 folder and that broke my setup, so that was a bad guess.

Would you have an idea where exactly this folder would be under Linux?

herrold, 2018-09-19:
@C0rn3j Your question really seems like something to answer by reviewing how Arch has packaged it, and then to solve the 'wrong' directory issue for libraries with a symlink

still packaging into ~/.auduino15/ for the base IDE seems ... very stale

C0rn3j, 2018-09-19:
@herrold Oh right, thanks, I read the package files and it's in /usr/share/arduino (I guess I could have just looked where the executable is pointing to).

@d-a-v I cloned the git master and pulled your changes as per your OP in the PR and I got this when trying to compile my project.

```
Build options changed, rebuilding all
In file included from /usr/share/arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/include/lwip/memp.h:58:0,
                 from /usr/share/arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/include/lwip/stats.h:43,
                 from /usr/share/arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/include/lwip/netif.h:50,
                 from /usr/share/arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/include/lwip/ip.h:46,
                 from /usr/share/arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:37:
/usr/share/arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/include/lwip/priv/memp_priv.h:48:32: fatal error: lwip/priv/mem_priv.h: No such file or directory
 #include "lwip/priv/mem_priv.h"
                                ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

d-a-v, 2018-09-19:
@C0rn3j sorry about that. The PR is updated

earlephilhower, 2019-01-16:
Closing as fixed by #5136

closed 2019-01-16 by earlephilhower

thrust26, 2020-02-25:
I recently got a product which uses the ESP8266 for WiFi connect. The Router only supports IPv6 and immediately deauthenticates the connections from the ESP8266 (Reason-Code 3).

Any idea what's wrong here?

d-a-v, 2020-02-25:
@thrust26 Please open a new issue with all the details you can provide by filling the issue template.

thrust26, 2020-02-25:
After a bit more research I am pretty the firmware does not support IPv6. Ignore my post above.