esp8266/Arduino #640: You leave no way to go back to earlier releases

craigk46, 2015-08-01:
How can I give my Arduino IDE an earlier release board manager URL?
I had code working perfectly and updating the board manager breaks my code.

In the future please leave a path to what has existed in prior releases so we can debug what we see changing in the behavior and decide what needs to be fixed.

I do appreciate everything you do. Now I'm very dependent upon your work.  
Thanks in advance.

igrr, 2015-08-01:
Thanks for feedback Craig, and sorry for the trouble.
You can go to the previous stable version using this URL:
http://arduino.esp8266.com/versions/1.6.4-673-g8cd3697/package_esp8266com_index.json

I will look for a way to merge multiple versions into a .json file automatically.

igrr, 2015-08-01:
Added package links to change log in 88834ac.

craigk46, 2015-08-01:
Many thanks.  I'm back to my code working better now but not perfectly.
I found if I removed the 1.6.5 from the board manager and then used the 1.6.4 URL that it was missing things and failed to compile.  I put 1.6.5 back in place, then using 1.6.4 that you provided it compiles fine.  I'm guessing that some of the 1.6.5 library stuff is still there because deep sleep is not working for me like it was before.  For now I'm just not going to use deep sleep until I figure out why it is not working like it worked before.

Thanks again for everything you do!

I'll submit a different issue later today regarding what I see broken in 1.6.5 relative to the Huzzah module.  I like the improvements in 1.6.5

craigk46, 2015-08-01:
False alarm about my problem with deep sleep.  I have it figured out and it was me not putting a jumper from gpio16 to rst on the Huzzah module.

Thanks again for giving the URL for your 1.6.4 version.

closed 2015-09-30 by igrr