esp8266/Arduino #641: Change SD Card

nemeier, 2015-08-01:
In some of my Projects I need to change the SD card. There is a simple allow this. I just require a single line in the SD.c library.

```
boolean SDClass::begin(uint8_t csPin) {
  /*
    Performs the initialisation required by the sdfatlib library.
    Return true if initialization succeeds, false otherwise.
   */
// ----> Line Added
  if (root.isOpen()) root.close();      // allows repeated calls
// <--- Line Added
  return card.init(SPI_HALF_SPEED, csPin) &&
         volume.init(card) &&
         root.openRoot(volume);
}
```

After changing the SD card all you need is to reinitialize the sd card.

Same Issue is posted in arduino as no 3607

gogela, 2016-09-20:
SD lib would definitely need a mechanism to re-read card as suggested or better to correctly close SD object implementing 'end()' method as opposite to 'begin()'

Interein2, 2016-11-23:
I've found this usefull, just because my SD card power and communication is mainly controlled by another mcu for logging. At sending data, host mcu releases lines, then ESP8266 module initializes SD, sends data and drops lines for the host to be able to communicate with SD card later. ESP restart in not an option.

sirzeta, 2018-02-27:
I created a PR #4444 that fixes this, is the same code applied here for this issue https://github.com/arduino-libraries/SD/issues/5