esp8266/Arduino #642: switching from Sta mode to Ap mode triggers wdt.

Duality4Y, 2015-08-01:
Hi,
When I switch the module from sta mode to ap mode the wdt is triggerd.
output:
```
mode is now AP_MODE

0.0.0.0
rm 0
pm close 7 0 0/10248092
del if0
usl
sul 0 0
mode : softAP(1a:fe:34:a4:4a:83)
add if1
dhcp server start:(ip:192.168.4.1,mask:25
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

```

but when i boot the device as access point first this does not happen and it works.

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004

Duality4Y, 2015-11-29:
will try!
If you don't mind me asking, what changed in the latest release that would make it go away you think?
just curious! :)

igrr, 2015-11-29:
Actually issue is still there. I was gdb-ing it just a few hours ago...

igrr, 2015-11-29:
Also, is this the same is #1073?

Duality4Y, 2015-11-29:
Actually this is pretty old,
Here it just resets by watchdog, while it that issue it crashes with a stack trace.