esp8266/Arduino #644: Turn Wifi on/off

raimohanska, 2015-08-02:
Is there an API for turning the WiFI off when it's not needed, to save power?

craigk46, 2015-08-02:
You can put this line at the end of your Arduino sketch to go to sleep for some number of microseconds.  In this case 25 seconds total.  The way it wakes from sleep is to have a gpio output pin pulse the reset pin of your module.  On the Adafruit ESP8266 Huzzah module you would tie gpio16 to rst so it can wake from sleep.  Realize this means a total reset so your module will run from powerup reset every time.  The nice thing is the module draws something measured in microamps while sleeping.

Another good reference for understanding this mode is here: https://www.sparkfun.com/news/1842

Below is the line of code that would work at the end of your Arduino sketch to have the module sleep for 25 seconds and wake from a cold reset if you tie the correct pin for your hardware to the RST input.

  ESP.deepSleep(25000000, WAKE_RF_DEFAULT);

raimohanska, 2015-08-02:
Thanks! I'm aware of the deep sleep feature though. I was just wondering if you could just turn off Wifi and have the processor run otherwise normally.

craigk46, 2015-08-02:
I don't know. I'm still learning about the ESP8266 myself.

torntrousers, 2015-08-03:
There is a thing called modem sleep metioned in the sdk doc which does what you want. The only way i've found to enable it though is by doing a deep sleep, eg ESP.deepSleep(1, WAKE_RF_DISABLED), and to re-enable wifi do another  ESP.deepSleep(1, WAKE_RF_DEFAULT). So not terrible convenient and if you need to keep track of the wifi state you need to store it in something like RTC memory. I see about 14 milliamps when running with wifi disabled compared with about 70 ma when its enabled but not sending much.

raimohanska, 2015-08-03:
Thanks @torntrousers, this is how it seems to me too. Just doesn't make sense that the only way to turn wifi on/off would be to go to deep sleep: ) Could this be something that the chip supports but is just missing from the Arduino API?

Links2004, 2015-08-03:
to disable WiFi use this:
```cpp
WiFi.mode(WIFI_OFF);
```
but i not now how many mA its save, would be interesting to know.

raimohanska, 2015-08-03:
@Links2004 yes there seems to be such an option, thanks! I made a Pull Request (#646) for including it in the docs. I didn't make power consumption measurements (yet) to see if it has any effect.

However, after doing `ESP.deepSleep(1000000, WAKE_RF_DISABLED)` the chip boots with Wifi disabled and calling `WiFi.mode(WIFI_AP_STA)` doesn't fix it. Resetting the chip after that will fix it though. But anyway, it doesn't seem possible to enable/disable WiFi on the fly.

torntrousers, 2015-08-04:
I hadn't heard of WiFi.mode(WIFI_OFF) either. Gave it a try measuring the change in current but i still get about 70milliamps after WiFi.mode(WIFI_OFF) whereas ESP.deepSleep(1, WAKE_RF_DISABLED) gives about 14milliamps on the wakeup, so it seems like WIFI_OFF does not completly switche off the wifi radio.

whogarden, 2016-01-12:
Hi,
Here is doc about WiFi.mode(WIFI_OFF) : http://esp8266.github.io/Arduino/versions/2.0.0/doc/libraries.html
But it is not working, and In a line in some sketch  if you put : WiFi.mode(WIFI_OFF); 
 WIFI_OFF is not recognized and stays in black color.
[image]

Links2004, 2016-01-12:
it will work.
the colore stuf comes from the IDE and the Arduino IDE is not intelligent when it comes to this.
add to to the keywords.txt
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/keywords.txt#L68
and it will be colored. 
a good IDE do this by itself, for arduino IDE its a manually process....

for more power saving use:
```cpp
WiFi.forceSleepBegin();
```
(only work in git or staring)

whogarden, 2016-01-12:
Yes it works,
Thanks a lot.

hallard, 2016-01-26:
Guys 
I'm enabling WiFi for 5 minutes to be able to set configuration of device, then I turn it off using 
```arduino
WiFi.mode(WIFI_OFF);
```
But it stop ESP from running and without any manual reset, my sketch is blocked. Looking inside WiFi code, `WiFi.mode` call SDK `wifi_set_op_mode()`
```arduino
void ESP8266WiFiClass::_mode(WiFiMode m)
{
    if(wifi_get_opmode() == (uint8)m) {
        return;
    }

    ETS_UART_INTR_DISABLE();
    if (_persistent)
        wifi_set_opmode(m);
    else
        wifi_set_opmode_current(m);
    ETS_UART_INTR_ENABLE();

}
```
and SDK 1.5.1 of 2016 page 49 on  `wifi_set_opmode` does absolutely not mention this mode off (0)
```
Parameters:
uint8 opmode: WiFi operating modes:
 0x01: station mode
 0x02: soft-AP mode
 0x03: station+soft-AP
```
Maybe calling this on `setup()` (before Wifi connect) works but in my case it doesn't
Here my code
```arduino
    // Time to Stop Wifi ?
    if (seconds >= MAX_WIFI_SEC ) {
      if (!(config.config & CFG_WIFI)) {
        DebugF(" Stopping Wifi ");
        webSocket.disconnect();
        yield();
        server.close();
        yield();
        WiFi.mode(WIFI_OFF);
        yield();
      }
    }

    DebuglnF("OK!");
    Debugflush();
```
and serial when it happens
```
task_1_sec... Stopping Wifi $T07#bb
```
Then need to push reset on ESP, you can see "OK!" string does not appears

Something I missed ?

igrr, 2016-01-26:
`$T07#bb`? You seem have GDBStub enabled? If so, perhaps you could fire up GDB and see what's going on?

chaeplin, 2016-01-26:
@hallard 
user_interface.h has 0 as NULL_MODE / https://github.com/esp8266/Arduino/blob/master/tools/sdk/include/user_interface.h#L146-L149

Examples in SDK 
```
wifi_station_disconnect(); 
wifi_set_opmode(NULL_MODE);
```

torntrousers, 2016-01-26:
I've tried the code from this comment http://www.esp8266.com/viewtopic.php?p=39624#p39624:
```
WiFi.mode(WIFI_OFF);
WiFi.forceSleepBegin();
delay(1); //Needed, at least in my tests WiFi doesn't power off without this for some reason
```
and it does switch WiFi off and the current use drops to about 14mA.
I've not though been able to successfully switch it on again. Doing:  
```
  WiFi.forceSleepWake();
  delay(1);
  WiFi.mode(WIFI_STA);  
  WiFi.begin(ssid, password);
```
the current use goes back up to about 70mA but it never successfully connects again. 

Can anyone show working code for how to switch on and connect again?

chaeplin, 2016-01-26:
@torntrousers Using example in SDK, I have 15mA, and wifi connection.

https://gist.github.com/chaeplin/be4d9ca41a991aa327bb

torntrousers, 2016-01-26:
Thanks @chaeplin, it looks like its the wifi_station_connect() in your code that gets it connected again, so this works for me:
```
  WiFi.forceSleepWake();
  WiFi.mode(WIFI_STA);  
  wifi_station_connect();
  WiFi.begin(ssid, password);   
```
I wonder shouldn't the wifi_station_connect call be included within the WiFi.begin function?

chaeplin, 2016-01-26:
@torntrousers WiFi.begin has wifi_station_connect already. https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L146

hallard, 2016-01-26:
@igrr 
Yes I enabled GDBStub to resolve some exception, but unfortunately, the file ` xtensa-lx106-elf-gdb` is not present on archive
`http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz`
So I can' go further, do you have any build for windows ?
I will try using 
```
WiFi.disconnect(); 
WiFi.mode(WIFI_OFF);
WiFi.forceSleepBegin();
delay(1);
````
as soon as I solve my GDB problem

hallard, 2016-01-26:
I grabbed gdb from gnutoolchains [here](http://gnutoolchains.com/download/) for those who're interested. I'm wondering If I can replace the whole ESP8266 toolchain bin directory from Arduino with this one from gnutoolchains ?

And following code did the trick (I don't need to enable it back so I didn't tested this part). make sense to call disconnect before off ;-)

```arduino
WiFi.disconnect(); 
WiFi.mode(WIFI_OFF);
WiFi.forceSleepBegin();
delay(1);
```

Thanks to everyone helping me

torntrousers, 2016-01-26:
@chaeplin interesting, but I can't get it to connect again without the explicit call to wifi_station_connect in my code. Does it work with/without for anyone else?

chaeplin, 2016-01-26:
@torntrousers Without wifi_station_connect, I can't also.

vachhaninimit, 2016-02-26:
Hi,

I have used WiFi.mode(WIFI_OFF) which turned off my wifi modem. But after that i am creating Access point with same chip but not able to find the SSID.  Is there any way to reset the chip or turn on the radio of wifi ?

gosewski, 2016-03-03:
For me it works without explicit call to wifi_station_connect. In my code I run the AP to get from user the SSID and PASS needed for STA mode, afterwards I switch mode to STA. Here is a part of my code to turn off the AP and turn on the STA on the fly:

```
...
client.stop();
server.stop(); 
WiFi.disconnect();
WiFi.mode(WIFI_OFF);
WiFi.mode(WIFI_STA);
WiFi.begin(chSSID, chPWD);
while (WiFi.status() != WL_CONNECTED) delay(500);
...
```

vachhaninimit, 2016-03-04:
@gosewski : Thanks for the info. I will give it a try and let you know.

matthuisman, 2016-04-17:
So, to confirm 

after:
ESP.deepSleep(1000000, WAKE_RF_DISABLED),

The only way to get WiFi working is
ESP.deepSleep(1, WAKE_RF_DEFAULT)

Would doing this have the same affect (just as low current use on next boot):

WiFi.disconnect(); 
WiFi.mode(WIFI_OFF);
WiFi.forceSleepBegin();
delay(1);
ESP.deepSleep(1000000, WAKE_RF_DEFAULT);

battika, 2016-04-23:
@matthuisman thanks for sharing this. I had exactly the same problem with Wi-Fi staying disabled after deepSleep and I was looking for an elegant way to turn it back on. Your second suggestion seemed to help. Thanks again.

drmpf, 2016-09-11:
void stopWiFiAndSleep() {
  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
  WiFi.forceSleepBegin();
  delay(1);
}

Gives me about 30mA supply current, down from 80-90mA  (WiFi running but not connected)
However after turning on an AP and handling the config I find I need to call this a couple of times to get back to 30mA.
(Perhaps a delay in sending last page is holding it up?)

drmpf, 2016-09-11:
Tried without the delay(1),  did not seem to work.  Also tried just one call after delay(5000), also did not work. So now I am calling it very loop. i.e.
```
void loop() {
  if (webserverStarted) {
    dnsServer.processNextRequest();
    webserver.handleClient();
    return;
  }

  stopWiFiAndSleep();
 // ... other normal processing
}

```
This seems to work (at least for the last few tests)

devyte, 2017-10-20:
The original issue is answered. Closing.
If there is any other issue pending from the discussion, please open a new issue, add the requested info, and reference the relevant comment(s).

closed 2017-10-20 by devyte