esp8266/Arduino #645: Support to store files locally on the ESP8266

mazloumi, 2015-08-02:
Hi, I was wondering if there is support to store files on the ESP itself like it is possible in the NodeMCU firmware. Interesting enough I first used the NodeMCU firmware and uploaded some files. Then I flashed the device with an application I wrote in Arduino. Then I went back to NodeMCU and the files I uploaded were still there. I assume different address spaces are used? How much space has the ESP to store firmware and other files? Can we support files with the Arduino IDE on the ESP itself?

igrr, 2015-08-02:
Yes, both storing and uploading are features in progress.

mazloumi, 2015-08-02:
When do you expect a release? Also, how much space is available on the ESP for firmware and other files?

far5893, 2015-08-05:
i suppose that in staging is supported spiffs

igrr, 2015-09-30:
Available in staging:
http://arduino.esp8266.com/versions/1.6.5-1160-gef26c5f/doc/reference.html#file-system

closed 2015-09-30 by igrr