esp8266/Arduino #651: Feature request: Allow for non-bootloader initiated serial based flash

CptanPanic, 2015-08-04:
Thinking about the OTA flashing, the nice thing is not to have to worry about making the various connections to get bootloader into flash mode.  So I was thinking that now that OTA is almost completed, maybe we can use the same mechanism to just flash code via serial also.  It seems that all that would have to be done is to adjust OTA example to listen to serial as well as UDP.  What do you think?

Duality4Y, 2015-08-07:
should be possible to use the underlying functions (the flash write and read functions) to flash over serial without restarting the device and boot it into bootloader mode. (that is what you meen right?)

igrr, 2015-08-07:
It is not possible to do this without restarting into bootloader mode.
Fortunately, Updater class is quite flexible — you can feed the new binary
from any source (network, serial, even from an SD card). So this should be
possible to do inside the sketch itself.

CptanPanic, 2015-08-07:
@igrr I don't quite understand your comment.  Your first sentence says it is not possible, but then you say it is possible.  So which is it?

igrr, 2015-08-07:
Doing this without restarting into bootloader (i.e. what was suggested by
Robert) is not possible. But using Updater class (which will restart to
bootloader) this is possible.

devyte, 2018-05-29:
Just took a look at the UpdaterClass. A Serial object can be directly used with it from a sketch or wrapper class.
Closing.

closed 2018-05-29 by devyte