esp8266/Arduino #653: esptool missing

jtheires, 2015-08-04:
On Arduino 1.6.4, I am setting up to program esp8211-01 modules through FTDI module from Windows PC through USB cable. Followed the installation instructions using boards manager (no issues so far). Went to select esptool as programmer in IDE and it's not listed in the menu item, even after a relaunch of IDE. Looked at discussions here and did not find any other issues logged. One post says esptool is automatic in 1.6.5, and does not appear in the tools/programmer menu, but no mention of 1.6.4.
Must esptool be selected in Arduino IDE 1.6.4 or not? If not, which programmer should be selected?

Juppit, 2015-08-04:
You don't need to select a programmer, just leave it what it is.
Esptool will be selected automaticlly.

jtheires, 2015-08-04:
Juppit,
Thanks, but I still get the same errors.
I tried both "ArduinoISP" and "Arduino as ISP" to no avail. Something has to be selected.
I was able to save once to the esp8266 but nothing since then.
There are many other options in the IDE (flash mode, flash frequency, etc), and I have seen no documentation on how to set these options.
Also, the ESP01_connect.jpg wiring diagram for FTDI to ESP had 2 ambiguous pins on the FTDI marked RTS and DTR, whereas my FTDI has DTD and CTS pins. I guessed and connected DTD to GPIO0 (instead of DTR) and CTS to RST/GPIO16 (which I assume is the same as EXT_RSBT in the diagram). At this point, the situation got worse, not better, as the ESP module refused to go into flash mode before attempting to program the ESP...

Juppit, 2015-08-04:
I do not have this FTDI, but you need TX, RX, CH_PD on Gnd. GPO0 must be 
on Gnd while power on, so the ESP will be in program mode.

You allready selected "Generic ESP8266 Module" on tools->board so you 
have only to select flash size 512K, all other items are ok by default.

jtheires, 2015-08-04:
Thanks Juppit.
Yes, I've been practicing the correct TX, RX, chipselect, 
GPIO0 connections since first learning of the 8266. The 
other 2 pins were news to me as of today.
I guess this means I have a flaky 8266, or perhaps a bad 
connection on my breadboard. Perhaps I'll spend some time 
trying to tighten up my connections - some guys swear by 
some spray for breadboards that foster better 
connections...haven't tried that yet.
James

Juppit, 2015-08-04:
James
I have to correct myself. CH_PD has to be connected to 3V3 not to Gnd, 
but you did so already...

KarthikR3, 2016-01-20:
hi iam getting this error while uploading to esp8266 using arduino ide 1.6.7
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte

closed 2016-02-29 by igrr

DMKhairnar, 2016-06-05:
follow esp8266 boar installation process 

For Program ESP8266
GPIO15=GPIO0= GND
EN = VCC

For Normal Execution Mod 
GIO15 = Float
GPIO0=GND
EN=VCC

and select board "Generic ESP8266 module"  dont change any settings
just select example and upload