esp8266/Arduino #655: Storing more than 4096

mc-hamster, 2015-08-05:
Good day,

For my application, I need to be able to store upwards of 128k. More would be nice.

I see that the eeprom library allows access to a sector of flash located at 0x7b000 for storage. Would it be feasible to have access to more by directing the eeprom library to use a different sector?

- Jm

igrr, 2015-08-05:
What is the size of flash chip you are using?
For 512k flash, you will likely have to implement that yourself (take a look at EEPROM library source code for ideas). For larger flash sizes there is a filesystem which you can use instead. I will write some documentation for it this week.

mc-hamster, 2015-08-05:
I have a few with 512k flash and a few with more. I'll take your comment that I should stay away from the 512k models. A 64k filesystem would be far too restrictive.

A filesystem if available would be great. I'll look forward to your documentation and provide feedback

Thanks!

- Jm

mc-hamster, 2015-08-11:
I found your unit tests at:
  https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/tests/FSWrapper/FSWrapper.ino

First pass works well. I'm working with your SPI library right now (I think I've found a bug, will file a report once I properly qualify it) and will move back onto this next.

If you're ever going to be in San Francisco, hit me up. We'll go for a beer. You do good work, it's appriciated.

Thanks!

- Jm

mc-hamster, 2015-08-14:
Closing this issue.

Thanks for the pointers.

- Jm

closed 2015-08-14 by mc-hamster