esp8266/Arduino #657: what are sleep mode.

shirish47, 2015-08-05:
can someone explain available sleep modes. I am not able to find any docs or reference of sleep modes.

AmrKafina, 2015-08-05:
You can find a list of official documents from espressif here: http://bbs.espressif.com/viewtopic.php?f=20&t=225

This document explains the three different sleep modes:
http://bbs.espressif.com/viewtopic.php?f=21&t=645

shirish47, 2015-08-05:
ok got it. 
Is there no way where I can put it into sleep(not necessarly deepsleep) and wake up by interrupts from say any 5 buttons connected to GPIO's? 
I want to make something like on press of button device wakes up and updates which button was pressed. I thought that was possible because GPIO can have individual Attachinterrupt functions... but I am not sure about making it sleep. I am sure deep sleep is not an option.

AmrKafina, 2015-08-05:
This should be doable by using the Modem-sleep mode, as only the wifi is turned off in that mode.

If using deep sleep, the only thing you can do is wake up the chip by triggering a low edge signal on the RESET pin. You can try directly connecting the GPIOs to the RESET pin, but I don't thing this will be practical in your case.

closed 2015-09-30 by igrr

mihino, 2016-01-16:
example of light sleep mode in arduino esp8266 2.0.0 version: LIGHT_SLEEP_T 
working code is:
```
// Required for LIGHT_SLEEP_T delay mode
extern "C" {
#include "user_interface.h"
}

...

    //WiFi.disconnect(); // DO NOT DISCONNECT WIFI IF YOU WANT TO LOWER YOUR POWER !
    wifi_set_sleep_type(LIGHT_SLEEP_T);
    delay(20000); // 20s
```

I used it in my [ESP8266 with DHT22 IoT client](http://homecircuits.eu/blog/esp8266-temperature-iot-logger/) project.

edmond1616, 2016-08-24:
how esp wake up from light sleep
and what does packet send from router contains for waking up esp from light sleep

rahuldeo2047, 2017-08-14:
I think as soon as delay time finishes. Please confirm that.
refer: https://community.blynk.cc/t/esp8266-light-sleep/13584

stoefln, 2018-03-14:
The documentation links are not working anymore, and I am in need for WiFiSleepType documentation. Can't find any!