esp8266/Arduino #659: EEPROM commit failure on Sparkfun Thing ((WRL-13231))

NetForces, 2015-08-05:
I'm trying to write to the EEPROM on the Sparkfun Thing (WRL-13231) (ESP8266 based). I wrote the following sketch:

```
#include <EEPROM.h>
 
void setup()
{
   unsigned char data;
   Serial.begin(115200);
   EEPROM.begin(10);
   data = EEPROM.read(0);
   Serial.println("Go");

   if(data == 0x55)
   {
      Serial.printf("Valid signature found, ...[0x%02x]\n",data);
   }
   else
   {
      Serial.printf("No signature found, writing...[0x%02x]\n",data);
      EEPROM.write(0,0x55);
      EEPROM.commit();
      delay(10);
   }
}

void loop() 
{
   Serial.println("Loop...");
   delay(1000);
}
```

On first boot it should write 0x55 at index 0 in EEPROM, then on subsequent boot if should always find 0x55 at index 0. But it does not happen, Seems like the 0x55 value is never written...

sticilface, 2015-08-05:
what happens if you change 
```
   EEPROM.begin(10);
```
to
```
   EEPROM.begin(512);
```

NetForces, 2015-08-05:
Same thing or worst sometimes I just get griblish on my terminal...

[image]

Then if I just comment-out the EEPROM.write, I get what I would expect as terminal output:

[image]

sticilface, 2015-08-05:
I got home, and managed to try your sketch that you've posted above, and it works totally fine.  
This leaves 
1) Your IDE
2) Your esp.

Are you using the staging version of the IDE / which version are you using? Do you have another ESP to try?

NetForces, 2015-08-05:
I'm using Arduino 1.6.5

I do have 2x ESP-01 modules. Will try to wire them up so I can bootstrap them with Arduino.

sticilface, 2015-08-05:
Sorry by IDE i meant the esp8266 version.  Do you have it installed by the boards manager?

NetForces, 2015-08-05:
Ok, so I had both http://arduino.esp8266.com/stable/package_esp8266com_index.json and https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json installed.

Removed both installed modules and only installed http://arduino.esp8266.com/stable/package_esp8266com_index.json

For the Sparkfun Thing (WRL-13231) I selected:

Flash module: DIO
Flash Freq: 80MHz
Upload using: Serial
Flash size: 4M (3M SPIFFS)

And now it seems ok.

First boot gives:

```
Go

No signature found, writing...[0xff]
Loop...
Loop..
```

Then subsequent reboot gives:

```
Go

Valid signature found, ...[0x55]
Loop...
Loop...
```

Which is what I expected.

igrr, 2015-08-05:
According to the [schematic](https://cdn.sparkfun.com/datasheets/Wireless/WiFi/SparkFun_ESP8266_Thing.pdf), SparkFun ESP8266 Thing uses a 512k flash chip, so it's a bit weird that 4M setting works for you :)

NetForces, 2015-08-05:
Hmmm I was thinking 4M was 4Mbit (512k), I'll try it with 512k

NetForces, 2015-08-05:
@igrr I really thing the Flash size drop down is in bits not bytes.

If I load the Sparkfun Arduino library for ESP8266 (https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/installing-the-esp8266-arduino-addon?_ga=1.23552444.784276653.1434647697), it auto-select a 4M flash in that dropdown for their 'Thing'.

igrr, 2015-08-05:
If you're not using the boards manager package from this project (i.e. using the one from SparkFun) then I'm not sure I can help you.

In this project I do use bytes for flash size menu options though.

NetForces, 2015-08-05:
I have now switch to this main project package instead of Sparkfun fork as it seems broken (our out of date).

So if you say that for a 4Mbit flash I should use 512K (64K SPIFFS) I fully trust you.

As it is now working as I expect, let's close this ticket.

Thank you for your support !

closed 2015-08-05 by NetForces