esp8266/Arduino #661: Underscores don't seem to be valid characters in access point ssid/passwords.

WRansohoff, 2015-08-06:
Using '_' characters in the ssid or password for an access point configured with WiFi.softAP seems to cause issues with the access point. I've noticed a few odd issues; when attempting to run a server on the access point, 'connection refused' is consistently returned from the only available address, and a port scan shows that port 80 is closed. My laptop is also unable to connect to the AP at all.

To reproduce, attempt to use the following code with and without '_' characters in the ssid and/or password:

```C
#include <ESP8266WiFi.h>

const char ssid[]     = "........";
const char password[] = "..........";

IPAddress addr = { 192, 168, 4, 1 };
WiFiServer server(80);

void setup() {
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(addr, addr, {255, 255, 255, 0});
  WiFi.softAP(ssid, password);

  delay(25);
  server.begin();
  delay(25);
}

void loop() {
  WiFiClient client = server.available();
  if (!client) { delay(500); return; }
  client.flush();

  String response = "HTTP/1.1 200 OK\r\nContent-Type: text/plain\r\n\r\nHello";
  client.print(response);
  delay(1);
}
```

Oddly enough, when you configure it as an open access point (no password), ssids with '_' characters seem to work fine. I think that it might have to do with how the access point config fields are populated in the softAP method, but I'm not sure:

    strcpy(reinterpret_cast<char*>(conf.ssid), ssid);

I also saw this behavior with '@', so other special characters may have issues too.

ErikBjare, 2016-03-20:
I managed to reproduce the issue.

I had a similar issue (connected ok, IP was assigned, but connection refused on HTTP requests) when trying to connect to another AP. In my case the special character that was the culprit was `-`, which works fine when making a soft AP, definitely weird behavior... 

I think this goes deeper than previously thought, my Particle Photon doesn't like my SSID either...

devyte, 2017-10-20:
@WRansohoff @ErikBjare is this issue still valid with latest git?

ErikBjare, 2017-10-20:
I don't have an ESP8266 available for testing right now, sorry.

devyte, 2018-09-06:
Closing due to lack of feedback and age.

closed 2018-09-06 by devyte

richard42, 2020-05-15:
This is still a current problem. I have an ESP8266 device and I built and installed the ZiModem firmware with the Arduino IDE, and this device will not connect to any WiFi networks with underscores in their SSIDs.

devyte, 2020-05-15:
Then I suggest working with the Zimodem authors to reduce to a MCVE that shows the issue with pure core code.

richard42, 2020-05-27:
I did write a very short program to demonstrate the problem, but then found that I could connect to my networks with underscores. So I tried again tonight with the zimodem software and found that I was able to connect to networks with underscores with this as well. I don't know what happened in my testing environment such that it didn't work previously, but does now. I have to assume it was just radio interference, weak antennas, and coincidence.

devyte, 2020-05-27:
Underscore is not a valid char for ssid per specification. Whether that is enforced is dependent on the wifi firmware of your wifi device.
That means it may work with one device, may not work with another, and may work sometimes with another.

richard42, 2020-05-28:
I don't think that's correct.  The spec ([802.11-2007](https://www.iith.ac.in/~tbr/teaching/docs/802.11-2007.pdf)) defines the SSID field in section 7.3.2.1 as a variable length field containing between 0 and 32 bytes. There is no character set defined, nor any constraints on the values of the bytes. As [this stackoverflow post](https://stackoverflow.com/questions/4919889/is-there-a-standard-that-defines-what-is-a-valid-ssid-and-password) indicates, "a 32-byte string of NUL-bytes is a valid SSID".

devyte, 2020-05-28:
Ah yes, I was thinking of hostname, and a case where the ssid and hostname were the same string. To be clear, the underscore limitation applies to hostname and not ssid. Apologies for the confusion.