esp8266/Arduino #662: Error in the libaries

Dylantje, 2015-08-06:
Dear all...
I try to do my best...
Only this error is killing my fun..
I cannot find a solution so please :+1: 

<code>De schets gebruikt 347.444 bytes (80%)  programma-opslagruimte. Maximum is 434.160 bytes.
Globale variabelen gebruiken 56.252 bytes (68%) van het dynamisch geheugen. Resteren 25.668 bytes voor lokale variabelen. Maximum is 81.920 bytes.
Uploading 351584 bytes from C:\Users\ADMINI~1\AppData\Local\Temp\build8750310664445775590.tmp/ESPEasy.cpp.bin to flash at 0x00000000
...warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
Ongeldige bibliotheek gevonden in C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266: C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266
Ongeldige bibliotheek gevonden in C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266: C:\Users\Administrator\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266
</code>

Thanks..

igrr, 2015-08-10:
Possible duplicate of #347, #97, #444.

devyte, 2017-10-20:
Closing per #3655 and due to age.

closed 2017-10-20 by devyte