esp8266/Arduino #663: Is there a simple Wifi setup example?

Protoneer, 2015-08-06:
I am working on a project that includes an ESP8266 and I was wondering if there was some kind of library that made it easy to configure the Wifi settings on the ESP8266. 

An example is https://github.com/jeelabs/esp-link . It starts out as AP and exposes a webpage that allows you to configure it and add it to exciting Wifi networks very easily. If the ESP8266 for some reason looses access the configures Wifi network it will revert back to an AP and can be reconfigured.

This makes the level of entry so much lower cause it can all be done with a mobile phone instead of having to program it.

luc-github, 2015-08-06:
This is what I did : https://github.com/luc-github/ESP8266
project is still on going but the part you are looking for is done 
[image]
[image]
[image]
[image]

Protoneer, 2015-08-06:
Thanks @luc-github . That is much more than what I was looking for. :) 

Looks pretty good and I am sure I will be able to use it.

closed 2015-08-06 by Protoneer

Duality4Y, 2015-08-07:
I am working on something similar, my code also when it can't connect, boots into AP mode and exposes a web-page, but the web-page is also exposed in station mode :)

tzapu, 2015-08-27:
and my take on it here https://github.com/tzapu/WiFiManager

drmpf, 2015-08-27:
My version is here http://www.forward.com.au/pfod/CheapWifiShield/index.html

In my case I expect an external pushbutton, pressed on powerup, to enter config mode.
My version also allows you to set the IP address and portNo of the ESP8266 if running as a server.
Also has support for configuring the settings of ESP8266 as a client connecting to cloud server, optionally with login credentials.

tzapu I like your idea of going into config mode if cannot connect.  Will think about that.

I have an idea for retrofitting house lighting with remote control via ESP8266 wifi using small components in a snake that you can push back up into the roof space after wiring up.  The standard wall switch would still work in parallel with the wifi control.  No new wiring needed just reconfigure the existing wiring.

My issue is that with all the hardware in the ceiling, my pushbutton config is difficult to use and also don't really want to turn the house power on and off to get into config mode.  So perhaps, using your idea, just turn off the router, but that would put all the lights into config mode.

Another idea which I need to test out is a reed switch and a very strong magnet, Strong enough to work through the ceiling plaster board.

Or some other magic wand .....

tzapu, 2015-08-27:
@drmpf don t forget to add some lost connection checks and some timeouts. i have some modules without that that, after a power cut, just get stuck in config mode and do nothing.

drmpf, 2015-08-27:
@tzapu  Don't seem to have that problem in my current version of the code, because it does not go into config mode until turned off and pushbutton held down while power applied.  
Then, in my code, after config via web page and getting confirmation webpage, you need to power cycle the ESP8266 to start up in normal mode (while NOT pressing the push button)
Bullet proof so far.

duncan-a, 2015-08-27:
@drmpf <https://github.com/drmpf> I like your WiFi setup method and the
idea for 'replacement wall switches' and it's something I've been puzzling
over.

I don't know where you're located, be here in Europe there is usually only
'Fixed Live' and 'Switched Live Return' at the light switch (i.e. there is
no neutral). The 'Switched Live Return" will be a "Virtual Neutral via the
bulb filament when the wall switch is in the off position, but both leads
will be at the same potential when the wall switch is on - so there's no
obvious method for powering any form of remote control device.

Contact-wise, substituting a two-way (changeover) switch for a one way
(on/off) and using a changeover relay for the remote control achieves
everything for a single-point light (it's more complex with multi-point
switching but is still achievable) - but it's no use if the remote control
device (ESP8266 or whatever) can't be powered continually.

I realise this is 'off topic' for a thread about WiFi setup but you seem to
have an idea of what you're doing so I thought I'd take the risk...

Back on topic (sort of), the 'reed switch and magnet' approach could be a
simple, but effective, solution - maybe flash the controlled light three
times to indicate configure mode is entered (if that's possible - it may
require another device, don't know, haven't though it through).

drmpf, 2015-08-28:
Maybe we could take this off-line for a more detailed discussion of retrofitting house lights.
Send me an email via pfod.com.au support. I think my solution fits your description.