esp8266/Arduino #670: WiFi.RSSI() doesn't work

kevin789900, 2015-08-07:
Any chance this will be supported?

Links2004, 2015-08-07:
check if you have the latest version.
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp#L385
support is added since 25 May.

Juppit, 2015-08-07:
Stable Readme still says (ver. 1.6.5-947-g39819f0, built on Jul 23, 
2015) "WiFi.RSSI() doesn't work", also Staging version (ver. 
1.6.5-990-gc8a63ce, built on Aug 03, 2015)

Upps: still in the Documentation for latest development version 
(https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md)

igrr, 2015-08-07:
Duh... forgot to edit this out from documentation. Sorry.

kevin789900, 2015-08-10:
New to github, looks like I have this version of the library as delivered through the Arduino IDE:  1.6.4-673-g8cd3697  Is the library updated to include the RSSI?  How would I tell that via github (sorry for newbie questions)?

closed 2015-09-30 by igrr