esp8266/Arduino #671: Description	Resource	Path	Location	ECLIPSE: Type 'CLOSED' is ambiguous ' Candidates are: '	ClientContext.h	line 182

far5893, 2015-08-08:
wl_definitions.h
lwip/tcp.h

Eclipse parser find and error, only ide  parser, compile is fine

igrr, 2015-08-08:
Perhaps it would make sense to file a bug against Eclipse parser since the
code compiles fine?

far5893, 2015-08-08:
Or arduino plugin!!

Links2004, 2015-08-08:
have you something named ```CLOSED``` in your code?
the latest git show me only warnings:
 - reorder
 - sign-compare
 - unused variable
 - unused function

far5893, 2015-08-10:
No but i suppose that there is something wrong in included library chain , compiler use makefile but i feel that  ide use another logic to parse code

Links2004, 2015-08-11:
yes build is makefile (created by eclipse), but for some strange reason the indexer sometimes do something different from the makefiles.

Links2004, 2015-11-27:
see: #525

devyte, 2017-10-20:
@far5893 @Links2004 this issue is ancient. Is it still valid?

far5893, 2017-10-20:
On eclipse oxygen and last slober arduino plugin  seem resolved.

closed 2017-10-20 by far5893