esp8266/Arduino #673: Can't use Travis CI to build ESP8266 Arduino sketches

marcoschwartz, 2015-08-08:
Hello,

I have a bunch of projects using this library that I would like to integrate into Travis CI. However I am having an issue with the integration, I think I am doing something wrong in the configuration file. Here is the file:

```
language: c
before_install:
  - "/sbin/start-stop-daemon --start --quiet --pidfile /tmp/custom_xvfb_1.pid --make-pidfile --background --exec /usr/bin/Xvfb -- :1 -ac -screen 0 1280x1024x16"
  - sleep 3
  - export DISPLAY=:1.0
  - wget http://downloads.arduino.cc/arduino-1.6.5-linux64.tar.xz
  - tar xf arduino-1.6.5-linux64.tar.xz
  - sudo mv arduino-1.6.5 /usr/local/share/arduino
  - sudo ln -s /usr/local/share/arduino/arduino /usr/local/bin/arduino
install:
  - arduino --pref "boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json" --save-prefs
  - arduino --install-boards esp8266:esp8266 --save-prefs
  - arduino --install-library "DHT sensor library"
script:
  - arduino --verify --board esp8266:esp8266:generic $PWD/cloud_data_logger/cloud_data_logger.ino
  - arduino --verify --board esp8266:esp8266:generic $PWD/dht_test/dht_test.ino
notifications:
  email:
    on_success: change
    on_failure: change
```

And here is part of the result on Travis CI:

```
$ arduino --install-library "DHT sensor library"
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Xlib:  extension "RANDR" missing on display ":1.0".
Initializing packages...
Preparing boards...
Downloading libraries index... 
Downloading libraries index... Downloaded 0kb of 58kb.
Downloading libraries index... Downloaded 10kb of 58kb.
Downloading libraries index... Downloaded 13kb of 58kb.
Downloading libraries index... Downloaded 23kb of 58kb.
Downloading libraries index... Downloaded 33kb of 58kb.
Downloading libraries index... Downloaded 38kb of 58kb.
Downloading libraries index... Downloaded 46kb of 58kb.
Downloading libraries index... Downloaded 56kb of 58kb.
Downloading libraries index... Downloaded 58kb of 58kb.
Updating list of installed libraries
Downloading library: DHT sensor library 
Downloading library: DHT sensor library Downloaded 0kb of 4kb.
Downloading library: DHT sensor library Downloaded 2kb of 4kb.
Downloading library: DHT sensor library Downloaded 3kb of 4kb.
Downloading library: DHT sensor library Downloaded 4kb of 4kb.
Verifying archive integrity...
Installing library: DHT sensor library
Updating list of installed libraries
2.41s$ arduino --verify --board esp8266:esp8266:generic $PWD/cloud_data_logger/cloud_data_logger.ino
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Xlib:  extension "RANDR" missing on display ":1.0".
Initializing packages...
Preparing boards...
Verifying...
cloud_data_logger.ino:2:25: fatal error: ESP8266WiFi.h: No such file or directory
compilation terminated.
Error compiling.
The command "arduino --verify --board esp8266:esp8266:generic $PWD/cloud_data_logger/cloud_data_logger.ino" exited with 1.
```

Can anyone help? Thanks!

igrr, 2015-08-08:
First step would be to pass verbose option to Arduino to see compiler
command line.

marcoschwartz, 2015-08-08:
Thanks for the fast answer! I just did that, here is the link:

https://travis-ci.org/openhomeautomation/esp8266-cloud/builds/74692535

igrr, 2015-08-08:
Arduino says it uses DHT library, but doesn't mention ESP8266WiFi library.
For some reason it didn't pick it up from the sketch.

marcoschwartz, 2015-08-08:
I think I made some progress: I put some " " around the library instead of <> and it doesn't complain about the missing library anymore. However now I got:

```
/home/travis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file {build.flash_ld}: No such file or directory
```
The complete log is at:

https://travis-ci.org/openhomeautomation/esp8266-cloud/builds/74704581

Any idea on how to fix this build.flash_ld issue? Thanks

igrr, 2015-08-11:
Your script resembles what I saw in Adafruit's repository [here](https://github.com/adafruit/travis-ci-arduino/blob/master/install.sh), except that you use sudo. Perhaps you could try to installing Arduino into your home directory?
Regarding the actual error, `cannot open linker script file {build.flash_ld}: No such file or directory`, I guess for some reason this build variable didn't get expanded by Arduino. Not sure what is the exact reason, haven't seen an error like this before. Normally flash_ld is set for each board variant. Might be something wrong with board selection (although I can't point at the exact issue).

marcoschwartz, 2015-08-11:
Thanks for the link to the Adafruit file, this made it work! Can't explain why yet, but thanks for the help.

closed 2015-08-11 by marcoschwartz

sej7278, 2016-06-07:
did we ever get to the bottom of why ```{build.flash_ld}``` isn't getting expanded, i'm still seeing it in today's master, i can't move on from 2.1.0