esp8266/Arduino #674: Moden sleep doubt

emilianocasas, 2015-08-08:
Hi, I am trying to put the esp to sleep unless it recieves something from a server. Then it wakes up and processes the message.
I read about "modem_sleep" which seems to be the solution of my problem. 
How can i make this call? Is it ok like this ?

ESP.deepSleep(WAKE_RF_DISABLED)

Or should I specify another mode?
I did not specify the time parameter, because the data that comes from the server is asynchronous.
Thanks and sorry for my english.

pgollor, 2015-08-10:
Hi,
please read this document.
http://bbs.espressif.com/viewtopic.php?f=21&t=645

The esp8266 is using the modem sleep mode automatically.

emilianocasas, 2015-08-11:
Hi, thanks for the help. I read that document and I did not realize. So if
my esp needs to be connected to a server full time, it will "sleep" until
it recieves info?? This would be great.

pgollor, 2015-08-11:
Hi,
i am not sure, but i think so.
I recognise this could only be working in station mode and not in soft ap mode.

closed 2015-11-11 by igrr