esp8266/Arduino #675: WDT Timeout on WIFI? or MQTT leak?

scargill, 2015-08-08:
So - simplest example of MQTT listening using PubSubClient... adding nothing more than actually serially printing the incoming string. Works for hours now problem.

But then I lost the WIFI - service provider issue - after maybe 10 minutes and one router reboot it was all back on. All my boards came back to live - except for the experimental ESP8266-Arduino board..

After several attempts to reconnect to the WIFI.... it reconnected-  connected to the MQTT server and...... WDT reboot and DEAD. (stopped at the "wdt reset").  No idea how to get beyond that.

philenotfound, 2015-08-13:
I can somewhat reproduce that, posting a message every 5 seconds, listening on the ESP and printing the string. I even get sporadic WDT reboot without loosing WIFI. I have tried with PubSub v1.99, v1.99.1 and HEAD (as of today).

scargill, 2015-08-13:
I had not given it anywhere near heavy testing like that. If your code is simple and reliably fails wny not put it in here and let's see if we can narrow it down. I use C and the normal SDK but would really love to use this environment but for me the basic WiFi+mqtt must be rock solid. Have you tried printing out heap size to see if it is dropping?

philenotfound, 2015-08-13:
I had used it with some additional libraries, to output something on a few LED matrices and I have noticed some pixels that where out of line.. I'm suspecting some memory corruption. I'm running some slimmed down tests at home right now, to see how often it happens. Will report back after work..

philenotfound, 2015-08-13:
Here's the code I've used to test:
https://gist.github.com/philenotfound/6ce702eb575c321a6287

Haven't noticed a significant decline in free heap, still WDT reboots happen once in a while

scargill, 2015-08-13:
Yup repeatable - here's my version as I could not quite get yours to work - so mine sends a message to itself every minute - 

https://bitbucket.org/snippets/scargill/7RAg4

And SURE ENOUGH after being left along for maybe 40 minutes...

Incoming: hello world
Incoming: hello world
Incoming: hello world
Incoming: hello world

ctx: cont 
sp: 3ffea830 end: 3ffeaab0 offset: 01b0

stack>>>
3ffea9e0: 3ffe8b48 001d4955 3ffe9768 4020230a

3ffea9f0: 3fff47d0 0000000b 0000000b 00000000

3ffeaa00: 00000000 00000000 00000000 00000000

3ffeaa10: 00000000 00000001 3fff47b8 0000000d

3ffeaa20: 0000000d 3fff47d8 00000000 00000000

3ffeaa30: 3fff47f0 00000000 00000000 3fff4870

3ffeaa40: 00000005 00000005 3fff4888 0000000a

3ffeaa50: 0000000a 0000000f 3fff47b8 00000007

3ffeaa60: 00000007 40207564 402020dc 00000001

3ffeaa70: 00000005 3fff4798 0000000d 0000000d

3ffeaa80: 3fffdc20 00000000 3ffeab08 3ffeaadc

3ffeaa90: 3fffdc20 00000000 3ffeaad4 4020184a

3ffeaaa0: 00000000 00000000 3ffe9a90 40100378

<<<stack<<<
ets Jan 8 2013,rst cause:2, boot mode:(1,6)

ets Jan 8 2013,rst cause:4, boot mode:(1,6)

wdt reset

sticilface, 2015-08-13:
I'm using neopixelbus, webserver, and pubsubclient.  

I constantly get reboots after a few hours, sometimes min, when running quite heavy animation sequences.   despite stripping things down i've not got to the bottom of it. 

these issues could all be related, especially the 10us recommendation for nointerrupts().
https://github.com/esp8266/Arduino/issues/548

my show() command for neopixelbus takes 6ms for 200LEDs.. which is far longer than the 10us recommended.

scargill, 2015-08-13:
But in my test - there are no significant delays - any my use of the MQTT is basically just a doctored version of the example - and so you would expect it to work perfectly...

Also in my normal C code for ESP8266 (not using Arduino) I code for long strips of the WS2812B lighting and it takes WAY more than a few microseconds - and that never crashes - so my money is on either the WIFI code or the MQTT code.   I wonder how many others have experienced these issues?? Anyone out there?

folny, 2015-08-14:
Hi

Also, for a long time to solve problems with restarting the WDT also use the code with the library Neopixelbus and control 512 pixels by 4 GPIO outputs code works sometimes an hour or two and then restarts. I can not be somewhere in the core Arduino disable or completely remove WDT ?.

asetyde, 2015-08-14:
This wdt is emulated , it can be disable I think ... 
I hope a strong rebuild of esp main core because not work so good .. Wdt / udp wifi / gpio support .
I use nodemcu 1.0 12e

folny, 2015-08-14:
We also use now nodemcu 1.0 12e my code, I also tested the ESP-01, ESP-03 but still there were the same problems.

asetyde, 2015-08-14:
I only nodemcu 1.0  , I think it is best as quality/performance but as sparkcore or proton , esp lacks on software ... I hope more core work and stop new feature .. Ota etc

scargill, 2015-08-14:
Alright - the plot thickens here. I've done some more tests and now having realised an obvious silly in my code,  I cannot get this to fail - and in the process discovered I don't have the foggiest idea about shielding - you can have a laugh at my expense on the blog below. I'm going to give this a LONG test - but here's where I'm up to, up to now. At this point I'm somewhat more confident. http://tech.scargill.net/testing-esp8266-wifi/

At the time of heading off for a swim, my one-second message counter is up to 820.. let's see later today how it does. I'm fairly confident this will survive a power cut, a WIFI outage and an MQTT break. But only time will tell.

asetyde, 2015-08-14:
Peter you've never seen problems about use esp.restart and serial ? 
Sometimes I reboot sometimes not

scargill, 2015-08-14:
Clearly this needs testing and fixing…

asetyde, 2015-08-14:
Thx , hope soon a good stability .. New function can be after

philenotfound, 2015-08-15:
Peter, the suggested delay(10) after client.loop() indeed fixes my simple test program... 
Have you found out why the delay is needed?

scargill, 2015-08-15:
It works very well but isn’t 100% for me – but I’m glad it’s working for you – as I understand it, background processes are carried out during the delay….  Now if you tried that with an ESP using C normally, the WIFI would die…..  

 

Do let me know if you discover that it isn’t 100% - I hope it is.

 

Pete.

asetyde, 2015-08-15:
....Now if you tried that with an ESP using C normally, the WIFI would die…..  

What does it means ?

philenotfound, 2015-08-15:
That was also my understanding.. still it should run those background tasks between main loop() executions... This delay after client.loop() seems more like a workaround than a fix :-( There is still something weird going on.. I still get reboots and hangs in my main project, where i do SPI transfers every 50ms... Doesn't matter if i do them in a Ticker callback or in the main loop.

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004