esp8266/Arduino #676: Should  WiFi.mode(WIFI_STA) switch off the AP?

Protoneer, 2015-08-09:
I am trying to switch between STA mode and AP mode. To go to client mode I use the following code:
```
  WiFi.mode(WIFI_STA);
  WiFi.begin("XXXXXXX","XXXXXXX");
```
The WiFi.begin does the client connection but a few seconds later the AP also a pairs and it seems to run in mixed mode.

Is this normal?

Links2004, 2015-08-09:
yes ```WiFi.mode(WIFI_STA);``` goes to STA mode only. 
what version do you use. this behavior is changed 27 days ago.
see https://github.com/esp8266/Arduino/commit/6233977403509b7eea2e6c666cc7e0d6d8426c55
and #529

Protoneer, 2015-08-09:
@Links2004  from what I can see I am using the staging version for Aug 3. (I changed my Arduino setting to point to the staging version and removed the stable version before doing so)

Should I call softAPdisconnect first?

Protoneer, 2015-08-09:
@Links2004 you were spot on. I could not compile softAPdisconnect indicating that the move of repository did not work. Reinstalled and it is now work. THANKS!!!

closed 2015-08-09 by Protoneer