esp8266/Arduino #677: Java sockets Porblem

anteph, 2015-08-09:
Hi,

I've been working with ESP8266 and Python sockets for a while and everything works fine. If I create, send some data, receive the answer and then close the socket multiple times, it works fine. 

But recently I've tested the same thing with java sockets and after some connections the ESP starts to show problems. If I add a  sufficient delay between connections, it works well. I think it has something to do with the way Java handles sockets, since everything works with Python. 

I close the connection on both sides (ESP and Java program), so the problem isn't there.

I think the delay solves the problem because probably ESP8266 has some kind of timeout in its sockets.

The ESP8266 is working as server, with the WifiServer class.

Does anyone suspect about the cause of this behavior?

Thanks!

drmpf, 2015-08-10:
I am also having problems. Have not looked into it yet. Suspect it might be a 'half close' problem.  Was planing to investigate using Wireshark

anteph, 2015-08-10:
Please let me know if you find something. I've been doing a lot of tests with Python sockets and everything is working fine. 

Do you know what is the default timeout for the ESP8266 sockets and the maximum of connections allowed? 

My problem seems to be deterministic, if I send commands in a row without delay, ESP usually stops answering after a fixed number of messages. Then, after some time, it starts to answer again. Because of that I think you are right, it's probably some half close problem.

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004

HayTran, 2017-09-17:
Here is my code fix your problem. Have a nice day! I've also found this. Yola

`/* Create a WiFi access point and provide a web server on it. */

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

/* Set these to your desired credentials. */
const char *ssid = "Hay Tran";
const char *password = "hoinguoikeben";
WiFiServer server(4567);
WiFiClient client;

void setup() {
  Serial.begin(115200);
  initHardware();
  setupWiFi();
}

void loop()
{
  if (!client.connected()) {
    // try to connect to a new client
    client = server.available();
    digitalWrite(D0, HIGH);
  } else {
    // read data from the connected client
    byte data = 0;
    if (client.available() > 0) {
      data = client.read();
      Serial.println();
      Serial.print(data, DEC);
      Serial.println();
      digitalWrite(D0, LOW);
    }
    if (data != -1) {
      Serial.print("Replying to client: ");
      Serial.println(data, DEC);
      client.write(data+3);
    }
  }

  Serial.print("Client status: ");
  Serial.println(client.status());
}
void setupWiFi()
{
  Serial.println();
  WiFi.mode(WIFI_AP);
  Serial.print("Configuring access point...");
  /* You can remove the password parameter if you want the AP to be open. */
  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  /* Start TCP Server */
  server.begin();
}

void initHardware()
{
  Serial.begin(115200);
  pinMode(D0, OUTPUT);
  digitalWrite(D0, HIGH);
}`