esp8266/Arduino #681: HSPI + I2C not working together

Jorgen-VikingGod, 2015-08-10:
I have a short example, where I want to have an OLED display connected to ESP8266 (by I2C)
and also an RFID module (by HSPI)

Cabling is done like this https://github.com/Jorgen-VikingGod/ESP8266-MFRC522

But if I have both in use, the display is black and the RFID module works really unstable.
If I comment the source for the display out - the RFID is working perfect.
The same for the display; works fine, if I remove the cabling and source of the RFID module.

Could it be, that the drivers have some issues?
Thx, Jorgen

Below is my test demo source code of both...
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <SPI.h>
#include <Wire.h>
#include <MFRC522.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define OLED_RESET 16  // RST-PIN for OLED (not used)
#define OLED_SDA    4  // SDA-PIN for I2C OLED
#define OLED_SCL    5  // SCL-PIN for I2C OLED
Adafruit_SSD1306 display(OLED_RESET);

#define RST_PIN   15  // RST-PIN for RC522 - RFID - SPI - Modul GPIO15 
#define SS_PIN     2  // SDA-PIN for RC522 - RFID - SPI - Modul GPIO2 
MFRC522::Uid oldUid;
MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance

const char* ssid     = "YOUR-SSID";
const char* password = "YOUR-PASSWD";

const char* host     = "192.168.0.x";
const char* client1  = "192.168.0.1";
const char* client2  = "192.168.0.2";

WiFiClient client;

void setup() {
  Serial.begin(115200);
  delay(250);

  // initial SPI bus for RFID module
  SPI.begin();           // Init SPI bus
  mfrc522.PCD_Init();    // Init MFRC522
  
  // initial I2C bus and OLED display
  Wire.begin(OLED_SDA, OLED_SCL);
  Wire.setClock(400000);
  
  // by default, we'll generate the high voltage from the 3.3v line internally! (neat!)
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // initialize with the I2C addr 0x3C (for the 128x64)

  Serial.print("connecting");
  display.clearDisplay();
  // text display tests
  display.setTextSize(1);
  display.setTextColor(WHITE);
  display.setCursor(0,0);
  display.println("connecting...");
  display.display();
  
  WiFi.begin(ssid, password);

  int retries = 0;
  while ((WiFi.status() != WL_CONNECTED) && (retries < 10)) {
    retries++;
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  
  display.clearDisplay();
  display.setCursor(0,0);
  display.println("WiFi connected");
  display.println("IP address: ");
  display.print(WiFi.localIP());
  display.display();

  delay(5000);
}

int value = 0;

void loop() {

  display.clearDisplay();
  display.setCursor(0,0);
  display.println("Processing RFID");
  display.display();

// Look for new cards
  if (!mfrc522.PICC_IsNewCardPresent()) {
    Serial.println("!mfrc522.PICC_IsNewCardPresent()");
    delay(500);
    return;
  }
  // Select one of the cards
  if (!mfrc522.PICC_ReadCardSerial()) {
    Serial.println("!mfrc522.PICC_ReadCardSerial()");
    delay(500);
    return;
  }
  // only process new cards
  if (mfrc522.uid.uidByte == oldUid.uidByte && mfrc522.uid.size == oldUid.size && mfrc522.uid.sak == oldUid.sak) {
    Serial.println("mfrc522.uid.uidByte == oldUid.uidByte && mfrc522.uid.size == oldUid.size && mfrc522.uid.sak == oldUid.sak");
    delay(500);
    return;
  }
  Serial.println("woohooo");

// Show some details of the PICC (that is: the tag/card)
  display.clearDisplay();
  display.println("Card UID:");
  Serial.print("Card UID:");
  dump_byte_array(mfrc522.uid.uidByte, mfrc522.uid.size);
  Serial.println();
  display.display();
  // remember old card
  rememberCard();

  delay(2000);

#if 0
  delay(3000);
  ++value;

  Serial.print("connecting to ");
  Serial.println(host);
  
  // Use WiFiClient class to create TCP connections
  
  const int httpPort = 3000;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
  
  // We now create a URI for the request
  String url = "/vote/client/";
  if (value % 2)
    url += client1;
  else
    url += client2;
  /*
  url += "?private_key=";
  url += privateKey;
  url += "&value=";
  url += value;
  */
  display.clearDisplay();
  display.setCursor(0,0);
  display.print("Requesting URL: ");
  display.println(url);
  display.display();
  
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  delay(1000);

  display.println("response: ");
  Serial.println("response:");
  
  // Read all the lines of the reply from server and print them to Serial
  while(client.available()){
    String line = client.readStringUntil('\r');
    
    display.print(line);
    Serial.print(line);
  }
  display.display();

  delay(2000);
#endif  
}

// Helper routine to dump a byte array as hex values to Serial
void dump_byte_array(byte *buffer, byte bufferSize) {  
  for (byte i = 0; i < bufferSize; i++) {
    display.print(buffer[i] < 0x10 ? " 0" : " ");
    display.print(buffer[i], HEX);
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}

// Helper method to copy card information
void rememberCard() {
  oldUid.size    = mfrc522.uid.size;
  oldUid.sak     = mfrc522.uid.sak;
  for (byte i = 0; i < mfrc522.uid.size; i++) {
    oldUid.uidByte[i] = mfrc522.uid.uidByte[i];
  }
}
```

igrr, 2015-08-11:
There shouldn't be any interactions between SPI and I2C drivers (in the sense that i don't see how that would be possible). One is hardware-only, the other is software-only. Since both devices are slaves, and are not used at the same time, timing issues can probably be ruled out. I would check what the supply current requirements of both devices are — RFID and a display may consume a lot, not to mention the ESP itself.

Jorgen-VikingGod, 2015-08-13:
@igrr thx for the hint with the current - I try to measure this on weekend.
Perhaps this is the reason, that only on at the same time is working.

ramespark, 2015-08-15:
Same here. Trying to resolve it. Looks like one serial peripheral will work at a time.

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004