esp8266/Arduino #683: Data Security

bkrajendra, 2015-08-11:
Dear friends 
Is there any data security scheme possible with ESP8266.
As I know https is not possible in ESP due to Digital Certificate storage issue.
Is there any other way to secure communication between ESP and browser?
Please help.

drmpf, 2015-08-11:
Depends what sort of security you need.
SipHash Secure Challenge and Response for micro-devices (AVR / Arduino)
http://www.forward.com.au/pfod/secureChallengeResponse/index.html
Describes the light weight message hash that pfodApp (www.pfod.com.au) uses to secure against unauthorized access or modification of the messages.
The library is in C code and will run on the ESP8266.

bkrajendra, 2015-09-04:
Thanks @drmpf ,
I will definitely try this...
What I need is security while communicating over http.
As normally all http data can be intercepted by anyone in same network connected to router.
If a simple n lite security model is used to just achieve this, can solve this issue.
I was even thinking of Something like Shamir's Secret sharing Algorithm..

devyte, 2017-10-20:
https is possible thanks to axTLS, but it has a high RAM requirement of just under 20KB per incoming connection in addition to what your code needs.
Closing.

closed 2017-10-20 by devyte