esp8266/Arduino #684: ESPtool 0.4.6 no longer working

sticilface, 2015-08-11:
0.4.5 works fine.  0.4.6 does not flash my node-mcu 0.9.  
normally works under generic esp8266, but no longer does.  
Does not working selecting node-mcc board either. 

copying 0.4.5 into the place of 0.4.6 works immediately. 
Using on Mac. 

error:
```
v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.wchusbserialfd110
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build2735345419855494557.tmp/xyz.bin success
opening port /dev/cu.wchusbserialfd110 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
```

igrr, 2015-08-12:
I have changed the reset method for NodeMCU 1.0 from ck to nodemcu, but forgot to change it for NodeMCU 0.9. This is now fixed, thanks for noting.

sticilface, 2015-08-14:
I can confirm that using 1.0 board works.  so i'll close it.

closed 2015-08-14 by sticilface