esp8266/Arduino #685: Application crashing on calling delay()

Attil, 2015-08-12:
Hello,
I have a problem where an application crashes when I call delay() from a function. This function is stored in Flash memory, at the address 40238504.
I get this error:
```
Exception (2):
epc1=0x3fff4bc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff4bc8 depc=0x00000000

ctx: sys 
sp: 3fff3780 end: 3fffffb0 offset: 01a0
```
If I do not call delay, this function (which I'm debugging) restarts ESP8266-12 somewhere inside it. If this happens, I get no errors, no stack traces or anything similar, ESP just restarts like it was plugged off/on to power source.

Any suggestion about the cause of this problem?
Thanks!

igrr, 2015-08-12:
You shouldn't call delay() and other blocking functions from Ticker callbacks and GPIO interrupts, which is what you seem to be doing.
If this is not what you are doing, please post the sketch which you use to reproduce this issue.

Attil, 2015-08-12:
I'm not calling delay() from these functions.

I'm really sorry, but I cannot post the whole sketch right now. It's really big (which is why I had to use ROM instead of RAM for it). I'm using your library, but I'm not using Arduino IDE and compiling it using xtensa toolchain with scons.

What confuses me the most is  this works:

```
void a()
{
delay(1000);
b();
}

void b()
{
}
```

while this crashes the system:

```
void a()
{
b();
}

void b()
{
delay(1000);
}
```
While I believe the effect of both actions should be the same.

Which is why I am thinking maybe function b is placed wrong in the memory or something similar. Function A, which works flawlessly is at 0x40234B50, while function B is at 0x40238504.

Other issue I think may be the problem is a possibility of stack overflow. Is there any way to check if its happening? I tried setting `#define CONT_STACKSIZE 8192`, but it didn't help.

igrr, 2015-08-12:
Ok, the problem is that according to your log,

```
ctx: sys 
sp: 3fff3780 end: 3fffffb0 offset: 01a0
```
The crash happens in 'sys' stack, while delay function may only be called from continuation stack. But it's really hard to say what the exact issue is because your description is very vague. Did you make some changes to core_esp8266_main.cpp? Perhaps you can produce some minimal example which triggers this issue and which you can show?

Attil, 2015-08-12:
Thanks, it seems I did go over the continuation stack and into the system stack! Classic stack overflow!

I have changed entries for stacksize `#define CONT_STACKSIZE 8192` in cont.h and Arduino.h (had to change in both, I guess i have something wrong with my include order...) and everything works great! Suggestion about checking the stack really helped me out!

Hope you have a nice day.

Cheers!

closed 2015-08-12 by Attil