esp8266/Arduino #686: ESP8266WebServer doesn't have an end/stop method

julianh2o, 2015-08-13:
Shouldn't this class have an end() method publicly available?

For that matter, I don't see one in WiFiServer either.. Am I missing something or is there just no way of stopping a running server through this API?

igrr, 2015-08-13:
WiFiServer doesn't have a stop method, but you should be able to stop it by destructing the object through an assignment:
```c++
myServer = WiFiServer();
```
Same should work for ESP8266WebServer, although I have likely forgot to do some housekeeping in the destructor (like `if (_currentArgs) delete[] _currentArgs;`).

julianh2o, 2015-08-13:
Ah okay, the destructor. That makes sense. I'll poke at some of my code and see what the implications of simply using the destructor would be. If it looks strange or doesn't leave the chip in the state I want, perhaps I'll submit a pull request for adding an end method.

Thanks Igrr!

closed 2015-08-13 by julianh2o