esp8266/Arduino #688: esp8266 arduino ide download

nkostoe, 2015-08-14:
Hie, i am new to the arduino esp ide and from sources i expected to find a section where i can dowload a full installer for windows as seen in the image attached. please assist me on how i can get the windows installer for esp arduino ide
[image]

igrr, 2015-08-14:
Quoting readme, 

### Installing with Boards Manager ###

Starting with 1.6.4, Arduino allows installation of third-party platform packages using Boards Manager. We have packages available for Windows, Mac OS, and Linux (32 and 64 bit).

- Install Arduino 1.6.5 from the [Arduino website](http://www.arduino.cc/en/main/software).
- Start Arduino and open Preferences window.
- Enter ```http://arduino.esp8266.com/stable/package_esp8266com_index.json``` into *Additional Board Manager URLs* field. You can add multiple URLs, separating them with commas.
- Open Boards Manager from Tools > Board menu and install *esp8266* platform (and don't forget to select your ESP8266 board from Tools > Board menu after installation).

davidrib, 2015-08-14:
, can  tel-meIyou how i can get thecanthpakages updated? Remove and install again?

davidrib, 2015-08-14:
Sorry. 
Can you tel-me how can i  get thes pakages updated?
Remove and install again?
Tanks

nkostoe, 2015-08-15:
i have downloaded the Arduino1.6.5 ide and installed the esp8266 board but i can not find the esptool under the programmers and when i try uploading a sketch i get this error message

"Arduino: 1.6.5 (Windows 7), Board: "Generic ESP8266 Module, Serial, 80 MHz, 80MHz, DIO, 115200, 2M (1M SPIFFS)"

E:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IE:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0/tools/sdk//include -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IE:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266 -IE:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic C:\Users\21FB2E~1\AppData\Local\Temp\build670289431897611190.tmp\ESP_Blink.cpp -o C:\Users\21FB2E~1\AppData\Local\Temp\build670289431897611190.tmp\ESP_Blink.cpp.o 

xtensa-lx106-elf-g++: error: CreateProcess: No such file or directory
Error compiling."

Links2004, 2015-08-15:
check if E:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++ is existing

nkostoe, 2015-08-15:
yes the E:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++ exists

asetyde, 2015-08-19:
You can also download official arduino , put esp stable/staging json on preferences , go to board manager , install esp (at the end of list ) , and all ok (download this just libraries and overwrite arduino standard .. onewire)

closed 2015-11-27 by Links2004

ekorpelathehacker, 2017-04-28:
I was wondering how do i do this comment out a part of the mac vendor list in data.h.