esp8266/Arduino #689: 'FS' is already declared in this scope

mc-hamster, 2015-08-14:
Good morning,

I ran into an issue with the FS library in the Aug 10 Stage build.

--

FS.h:112:11: error: 'FS' is already declared in this scope
 using fs*:FS;
Error compiling

--

I've tracked down the issue to the order of which the ESP8266WebServer and FS libraries are included.

If they are included in this order:

#include &lt;ESP8266WebServer.h&gt;
#include "FS.h"

The above error message will block compilation.

If they're included in this order:

#include "FS.h"
#include &lt;ESP8266WebServer.h&gt;

There is no error message.

Thanks!

- Jm

igrr, 2015-08-14:
Should be fixed in 87d199fc (but not available in new staging build yet).

mc-hamster, 2015-08-14:
Thanks. I'll cherry pick your change when I get home from work.

- Jm

closed 2015-09-30 by igrr