esp8266/Arduino #691: WebServer crashing

mozgy, 2015-08-14:
Heya,
this code almost reliably gets WebServer crash - https://github.com/mozgy/esp8266/blob/master/Blink/WebServerBlink.ino
It's NodeMCU v0.9 and /staging/ -1044 version

Serial debug output ->
Heap: 37136
Boot Vers: 31
CPU: 80
SDK: 1.3.0
Chip ID: 10768655
Flash ID: 1458415
Vcc: 3002

AP IP address: 192.168.4.1
add 1
aid 1
station: c4:9a:02:13:b9:37 join, AID = 1
favicon.ico
Elapsed       29
LED ON
Elapsed       31
favicon.ico
Elapsed       35
favicon.ico
Elapsed       43
LED OFF
Elapsed       52
favicon.ico
Elapsed       58
favicon.ico
Elapsed     1:01
LED ON
Elapsed     1:21
favicon.ico
Elapsed     1:25
favicon.ico
Elapsed     1:27
LED ON
Elapsed     1:32
favicon.ico
Elapsed     1:36
favicon.ico
Elapsed     1:38
LED OFF
Elapsed     2:00
favicon.ico
Elapsed     2:04
favicon.ico
Elapsed     2:07
LED ON
Elapsed     2:12
favicon.ico
Elapsed     2:16
favicon.ico
Elapsed     2:18
LED ON
Elapsed     2:55
favicon.ico
Elapsed     2:59
favicon.ico
Elapsed     3:02
LED OFF
Elapsed     3:04
€
 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Heap: 37136
Boot Vers: 31
CPU: 80
SDK: 1.3.0
Chip ID: 10768655
Flash ID: 1458415
Vcc: 3000

add 1
aid 1
station: c4:9a:02:13:b9:37 join, AID = 1
AP IP address: 192.168.4.1
favicon.ico
Elapsed       19
LED OFF
Elapsed       25
favicon.ico
Elapsed       29
LED ON
Elapsed       31
favicon.ico
Elapsed       35
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
favicon.ico
Elapsed       48
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Heap: 37136
Boot Vers: 31
CPU: 80
SDK: 1.3.0
Chip ID: 10768655
Flash ID: 1458415
Vcc: 2998

add 1
aid 1
station: c4:9a:02:13:b9:37 join, AID = 1
AP IP address: 192.168.4.1
favicon.ico
Elapsed       24
LED ON
Elapsed       28
favicon.ico
Elapsed       32
favicon.ico
Elapsed       34
LED OFF
Elapsed       46
favicon.ico
Elapsed       50
favicon.ico
Elapsed       53
LED ON
Elapsed     1:02
favicon.ico
Elapsed     1:06
favicon.ico
Elapsed     1:08
LED ON
Elapsed     1:19
favicon.ico
Elapsed     1:24
LED ON
Elapsed     1:26
Fatal exception (9):
Fatal exception (9):
Fatal exception (9):
Fatal exception (9):
Fatal exception (9):
Fatal exception (9):
Fatal exceatal exception ( ):
Fatal exception (
 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Heap: 37136
Boot Vers: 31
CPU: 80
SDK: 1.3.0
Chip ID: 10768655
Flash ID: 1458415
Vcc: 2999

add 1
aid 1
station: c4:9a:02:13:b9:37 join, AID = 1
AP IP address: 192.168.4.1
favicon.ico
Elapsed       22
LED ON
Elapsed       25
favicon.ico
Elapsed       29
favicon.ico
Elapsed       31
LED OFF
Elapsed       33
favicon.ico
Elapsed       37
favicon.ico
Elapsed       45
LED ON
Elapsed       58
favicon.ico
Elapsed     1:03
favicon.ico
Elapsed     1:05
LED ON
Elapsed     1:10
favicon.ico
Elapsed     1:14
Fatal exception (9):
epc1=0x401018b9, epc2=0x00000000, epc3=0x00000000, excvaddr=0xd903a24a, depc=0x000000
Exception (9):
epc1=0x401018b9 epc2=0x00000000 epc3=0x00000000 excvaddr=0xd903a24a depc=0x00000000

ctx: cont 
sp: 3ffeaff0 end: 3ffeb360 offset: 01a0

>>>stack>>>
3ffeb190:  3ffeb200 40213405 00000000 00000000  
3ffeb1a0:  0000005a 00000000 4021588c 3fff3bc0  
3ffeb1b0:  00000000 00000000 000000ff 00000000  
3ffeb1c0:  00000336 3ffeb1e0 40215aad 3ffeb1f0  
3ffeb1d0:  00000001 00000001 3ffe9215 40208094  
3ffeb1e0:  3ffe0000 00000000 00000000 3ffe97f8  
3ffeb1f0:  3ffe9fc4 00000010 3ffe8428 00000000  
3ffeb200:  0000
000001 40203a76 3fff3800 00000336  
3ffeb210:  3ffe9fc4 3ffe97f8 00000336 00000336  
3ffeb220:  00000000 3fff35e0 00000000 00000001  
3ffeb230:  3ffe9fc4 3ffe979c 3ffeb3b8 40202692  
3ffeb240:  3ffe8e98 00000000 000003e8 0000000c  
3ffeb250:  3ffeb310 3fff3800 0000000c 3fff3660  
3ffeb260:  3ffe97c4 0000000c 3ffeb2b0 40208ace  
3ffeb270:  3fff3cc8 3fff3360 3fff3360 40207912  
3ffeb280:  3ffe9fc4 3fff3360 3ffeb2b0 402054b6  
3ffeb290:  40208a55 3ffe97c4 3fff3360 402054e6  
3ffeb2a0:  3fff3cb0 00000000 3ffe97c4 40205158  
3ffeb2b0:  3fff35e0 0000000c 0000000c 0000000c  
3ffeb2c0:  0000000c 3fff3608 00000003 00000003  
3ffeb2d0:  3fff3c80 00000019 00000019 40204229  
3ffeb2e0:  402054c0 00000001 000003e8 40203fb2  
3ffeb2f0:  00000000 3fff3800 3ffe97c4 3ffeb38c  
3ffeb300:  3fffdc20 000003e9 3ffe97c4 40205269  
3ffeb310:  3ffe8e98 00000000 000003e8 000122ed  
3ffeb320:  3ffe97dc 3fff3800 00000016 40101f6d  
3ffeb330:  4020180d 00000000 3ffeb384 4020222c  
3ffeb340:  00000000 00000000 3ffeb384 4020186a  
3ffeb350:  00000000 00000000 3ffea340 40100374  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Heap: 37136
Boot Vers: 31
CPU: 80
SDK: 1.3.0
Chip ID: 10768655
Flash ID: 1458415
Vcc: 3001

add 1
aid 1
station: c4:9a:02:13:b9:37 join, AID = 1
AP IP address: 192.168.4.1
station: c4:9a:02:13:b9:37 leave, AID = 1
rm 1

bardiir, 2015-09-07:
I got sort of a similar issue. Despite me doing a client.close() and only using local variables everywhere so it get's cleared up again once the request handling is done it does run out of heap after a while.
However after some time, it seems there is sort of a timeout or garbage collection running which makes heap come back slowly. So if the requests are slow enough in succession it can handle requests indefinitely in my case.

igrr, 2015-09-07:
@bardiir given the description, your issue is not related to this one.
In your case TCP connection ends up in TIME_WAIT state, where they are kept for 2 x MSL timeout, according to TCP spec. When timeout passes, resources owned by the connection are deallocated, and you see heap usage go down.

mozgy, 2015-09-07:
Tried latest staging, still crashing, but I have some interesting new info, I have LG G3 and with it's internal 'Internet' browser it crashes a lot but installed Chrome and apparently cannot make it crash.
```
Heap: 38104
Boot Vers: 31
CPU: 80
SDK: 1.3.0
Chip ID: 10768655
Flash ID: 1458415
Vcc: 3005

AP IP address: 192.168.4.1
add 1
aid 1
station: c4:9a:02:13:b9:37 join, AID = 1
LED OFF
Elapsed       16
favicon.ico
Elapsed       20
favicon.ico
Elapsed       22
LED OFF
Elapsed       24
Fatal exception (9):
Fatal exception (9):
Fatal exception (9):
Fatal exception (9):
Fatal exception (9):
Fatal exception (9):
Fatal exce
 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Heap: 38104
Boot Vers: 31
CPU: 80
SDK: 1.3.0
Chip ID: 10768655
Flash ID: 1458415
Vcc: 3006

AP IP address: 192.168.4.1
add 1
aid 1
station: c4:9a:02:13:b9:37 join, AID = 1
favicon.ico
Elapsed       21
LED OFF
Elapsed       23
favicon.ico
Elapsed       27
favicon.ico
Elapsed       35
LED ON
Elapsed       37
favicon.ico
Elapsed       41
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
Fatal exception (9):
epc1=0x40101802, epc2=0x00000000, epc3=0x00000000, excvaddr=0x296f6b63, depc=0x00000000

Exception (9):
epc1=0x40101802 epc2=0x00000000 epc3=0x00000000 excvaddr=0x296f6b63 depc=0x00000000

ctx: cont 
sp: 3ffeaec0 end: 3ffeb1f0 offset: 01a0

>>>stack>>>
3ffeb060:  3fff3cf8 401017d4 3fff3bc0 000000b9  
3ffeb070:  40209c88 3ffeb0f0 000000b9 000000b9  
3ffeb080:  40209c88 00000001 3ffeb0f0 40209cb6  
3ffeb090:  3fff3a98 00000000 3ffeb0f0 4020a07d  
3ffeb0a0:  3fff3a90 00000000 3ffeb0f0 3ffe964c  
3ffeb0b0:  0000000d 3ffeb190 3ffeb0f0 4020a0fc  
3ffeb0c0:  3ffe0033 3ffeb0f0 3ffeb190 402097d7  
3ffeb0d0:  40209c3d 3ffeb190 3ffeb0f0 40209bd6  
3ffeb0e0:  3ffe9ed4 3ffe94f0 3ffeb190 402068eb  
3ffeb0f0:  3fff3bc0 000000b8 000000b8 3fff3a28  
3ffeb100:  0000004a 0000004a 3fff3980 0000000a  
3ffeb110:  00000006 3ffe964c 3ffe9680 40209e20  
3ffeb120:  3fff3740 00000000 00000000 3fff24e0  
3ffeb130:  00000000 00000000 3fff3728 00000001  
3ffeb140:  00000001 3fff2e38 00000003 00000003  
3ffeb150:  3fff3a90 00000052 00000052 4020478d  
3ffeb160:  3ffe8be0 00000000 000003e8 00000006  
3ffeb170:  00000000 3fff37b0 3ffe964c 3ffeb21c  
3ffeb180:  3fffdc20 000003e9 3ffe964c 40205849  
3ffeb190:  3ffe8be0 00000000 000003e8 0000e5e5  
3ffeb1a0:  3ffe9664 3fff37b0 00000001 402010a3  
3ffeb1b0:  00000000 00000000 3ffeb214 40202951  
3ffeb1c0:  4020180d 00000000 00000000 3ffeb21c  
3ffeb1d0:  3fffdc20 00000000 3ffeb214 4020186a  
3ffeb1e0:  00000000 00000000 3ffea1d0 40100374  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Heap: 38104
Boot Vers: 31
CPU: 80
SDK: 1.3.0
Chip ID: 10768655
Flash ID: 1458415
Vcc: 3006

add 1
aid 1
station: c4:9a:02:13:b9:37 join, AID = 1
AP IP address: 192.168.4.1
station: c4:9a:02:13:b9:37 leave, AID = 1
rm 1
```

igrr, 2015-09-07:
This is the same issue as #428.

igrr, 2015-09-30:
Fix released in staging.

closed 2015-09-30 by igrr