esp8266/Arduino #692: Error message on uploading sketch

nkostoe, 2015-08-15:
i have downloaded the Arduino1.6.5 ide and installed the esp8266 board but i can not find the esptool under the programmers and when i try uploading a sketch i get this error message. i am using windows 8.1 os. How can i get around the problem?

"Arduino: 1.6.5 (Windows 7), Board: "Generic ESP8266 Module, Serial, 80 MHz, 80MHz, DIO, 115200, 2M (1M SPIFFS)"

E:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IE:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0/tools/sdk//include -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IE:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266 -IE:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\generic C:\Users\21FB2E~1\AppData\Local\Temp\build670289431897611190.tmp\ESP_Blink.cpp -o C:\Users\21FB2E~1\AppData\Local\Temp\build670289431897611190.tmp\ESP_Blink.cpp.o

xtensa-lx106-elf-g++: error: CreateProcess: No such file or directory
Error compiling."

Juppit, 2015-08-15:
In my installation I do not find a path like yours 
"\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\tools".

So I gues you didn't download arduino-1.6.5-r2-windows.exe from 
https://www.arduino.cc/en/Main/Software as Windows Installer.

When executing the downloaded file, it will install in
"C:/Program files (x86)/Arduino".

The Arduino IDE buildes my workspace in
"C:/User/[username]/AppData/Roaming".

Using cygwin under Windows 8.1 "Arduino IDE for ESP8266" is installed in 
my Cygwin user directory, this is:
"~/Arduino/".

It looks like, there is somthing wrong with your installation.

> "Arduino: 1.6.5 (Windows 7), Board: "Generic ESP8266 Module, Serial, 80
> MHz, 80MHz, DIO, 115200, 2M (1M SPIFFS)"
>
> E:\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++

You are using Windows 8.1, not Windows 7?

nkostoe, 2015-08-15:
yes i am using win8.1

nkostoe, 2015-08-15:
the thing is i am running the arduino ide from my flash, can that be the problem?

Links2004, 2015-11-27:
please retry in last release see:
https://github.com/esp8266/Arduino#installing-with-boards-manager

If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004

Rishi159, 2019-03-29:
xtensa-lx106-elf-g++: error: CreateProcess: No such file or directory
Also, the json file linked is possibly expired or unavailable because when you paste it in preferences it shows unable to download kinda message