esp8266/Arduino #698: I2C error

petrd, 2015-08-16:
Hello,
I tried to connect ESP-12E I2C humidity and temperature sensor AM2321. I2C communication is not working. I tried it on Modemcu 0.9 (ESP-12) and on Modemcu 1.0 (ESP-12E). The same code works with the Arduino Due. I reduced the pull-up resistors from 10k to 2k2. Also it is not working. I tried HTU21D, it is similar to the humidity and temperature sensor. Also not working.
I do not know what I'm doing wrong, but it looks like it is not working I2C?

Configuration:
SDA = GPIO4
SCL = GPIO5
Pull-up resistors 2x10K or 2x2k2
Version: 1.6.5-1044-g170995a

Code:
<pre><code>
#include &lt;Wire.h&gt;

#define I2C_ADDR_AM2321 (0xB8 >> 1)
#define PARAM_AM2321_READ 0x03
#define REG_AM2321_HUMIDITY_MSB 0x00
#define REG_AM2321_HUMIDITY_LSB 0x01
#define REG_AM2321_TEMPERATURE_MSB 0x02
#define REG_AM2321_TEMPERATURE_LSB 0x03
#define REG_AM2321_DEVICE_ID_BIT_24_31 0x0B

void setup() {
  Serial.begin(9600);
  //Wire.pins(4,5); // SDA, SCL  
  //Wire.setClock(100000);
  Wire.begin(4, 5);
}

void loop() {

        //
        // Wakeup
        //
        Wire.beginTransmission(I2C_ADDR_AM2321);
        Wire.endTransmission();

        //
        // Read Command
        //
        Wire.beginTransmission(I2C_ADDR_AM2321);
        Wire.write(PARAM_AM2321_READ);
        Wire.write(REG_AM2321_HUMIDITY_MSB);
        Wire.write(4);
        Wire.endTransmission();

        //
        // Waiting
        //
        //delayMicroseconds(1600); //>1.5ms
        delay(2);
        //
        // Read
        //
        Wire.requestFrom(I2C_ADDR_AM2321, 8); // COMMAND + REGCOUNT + DATA + CRCLSB + CRCMSB
        //int i = 0;
        uint8_t buf[6];
        for (int i = 0; i < 6; ++i)
            buf[i] = Wire.read();

        unsigned short crc = 0;
        crc  = Wire.read();     //CRC LSB
        crc |= Wire.read() << 8;//CRC MSB

       // if (crc == crc16(buf, i))
       //     return true;
       // return false;

        unsigned int humidity;
        int temperature;
        humidity     = buf[2] << 8;
        humidity    += buf[3];
        temperature  = buf[4] << 8;
        temperature += buf[5];

        Serial.print("Temperature = ");
        Serial.print(temperature);
        Serial.print("\t Humidity = ");
        Serial.println(humidity);
        delay(3000);
}
</code></pre>

igrr, 2015-08-16:
> #include <Wire.h>

Which pins on Nodemcu board are you using? You know that silkscreen labels
do not correspond to GPIO number right?

petrd, 2015-08-16:
Hello,
I use D1 and D2, it should correspond GPIO5 and GPIO4.

petrd, 2015-08-16:
I tried also GPIO12 and GPIO14, the result is the same.

igrr, 2015-08-16:
Ok, I don't have these sensors at hand, please attach oscilloscope traces
so that we can figure out what is the cause of this issue.

petrd, 2015-08-17:
Hello,
so for me does not work Wire.write ()

First shot of the Arduino UNO + AM2321:
[image]
[image]

The following is a picture ESP-12E + AM2321:
[image]
[image]

igrr, 2015-08-17:
I can't zoom the traces enough to see the pulses, but from the log it looks like the difference is that ESP doesn't generate STOP condition after a NACK.

igrr, 2015-08-17:
You can try to edit twi_writeTo function in core_esp8266_si2c.c. Make it call twi_write_stop on errors (i.e. before `return 2` statement).

petrd, 2015-08-17:
Sending an enlarged graph which corresponds to the following code:

<pre><code>
        //
        // Wakeup
        //
        Wire.beginTransmission(I2C_ADDR_AM2321);
        Wire.endTransmission();

        //
        // Read Command
        //
        Wire.beginTransmission(I2C_ADDR_AM2321);
        Wire.write(PARAM_AM2321_READ);
        Wire.write(REG_AM2321_HUMIDITY_MSB);
        Wire.write(4);
        Wire.endTransmission();

</code></pre>

Arduino Uno:
        Wire.beginTransmission(0xB8);
        Wire.endTransmission();
        Wire.beginTransmission(0xB8);
        Wire.write(0x03);
        Wire.write(0x00);
        Wire.write(0x04);
        Wire.endTransmission();

[image]

[image]

Arduino ESP-12E:
        Wire.beginTransmission(0xB8);
        Wire.endTransmission();
        Wire.beginTransmission(0xB8);
        Wire.write(0x03); // Error
        Wire.write(0x00); // Error
        Wire.write(0x04); // Error
        Wire.endTransmission();

[image]

[image]

igrr, 2015-08-17:
Okay, graphs show that ESP doesn't generate a STOP condition after the first NACK. Please try the edit suggested above — it might resolve the issue.

petrd, 2015-08-17:
Hello,
I edited core_esp8266_si2c.c and AM2321 works.
Thank you very much for your help.

igrr, 2015-08-17:
That's great! Could you please post the diff or the changed code for the
record? (just to be sure I won't mess up anything when doing this change).

petrd, 2015-08-17:
OK, Thank you very much.

closed 2015-11-08 by igrr

Rorschac, 2018-07-16:
Hi, Ivan.

I want to transform "ESP test code" to Arduino uno for CAN-BUS comunication. I removed specified ESP code from your code, and compiled for UNO with no errors. Then I send for example "X" on arduino serial interface. I can see the messages with scope but Attiny85 did not response this message. Where I am doing wrong. I added that removed codes from your code and scope screen.

[image]
```
/*extern "C"
{
  #include "user_interface.h"
}*/

//os_timer_t myTimer;

//  os_timer_setfn(&myTimer, timerCallback, NULL);
//  os_timer_arm(&myTimer, 1000, true);

Wire.setClockStretchLimit(1000);
```

commit 46468d4a (2015-11-08): I2C: generate STOP in case of NACK (fix #698, #254)

thanks @petrd and @Yazzcat
commit d1a0c538 (2015-11-10): Squashed commit of the following:

commit 810ab68ae5ad89d24a57ed8d18c8ff6acbb54349
Author: Ivan Grokhotkov <igrokhotkov@gmail.com>
Date:   Mon Nov 9 01:37:22 2015 +0300

    Fix WiFiClientSecure::connected (#43)

    thanks @whyameye

commit 7384c3814b780e747940d9735fe08825a26f746c
Merge: 46468d4 464d891
Author: Markus <Links2004@users.noreply.github.com>
Date:   Sun Nov 8 23:04:52 2015 +0100

    Merge pull request #985 from Links2004/docu

    add missing exit in php sample

commit 464d891877a3fb153844a0110ca865861815b063
Author: Markus Sattler <help.markus+git@gmail.com>
Date:   Sun Nov 8 2 [... cut]