esp8266/Arduino #700: Staging version compile error

asetyde, 2015-08-17:
"C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-942-gc6e2a29/tools/sdk//include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=165 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266 -I"C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-942-gc6e2a29\cores\esp8266" -I"C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-942-gc6e2a29\variants\nodemcu" -I"C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-942-gc6e2a29\libraries\EEPROM" -I"C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-942-gc6e2a29\libraries\ESP8266WiFi\src" -I"C:\Program Files (x86)\Arduino\libraries" -I"C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-942-gc6e2a29\libraries" -I"C:\Program Files (x86)\Visual Micro\Visual Micro for Arduino\Micro Platforms\default\debuggers" -I"C:\Users\asetyde\Documents\Arduino\libraries" "C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-942-gc6e2a29\libraries\ESP8266WiFi\src\WiFiUdp.cpp" -o "C:\Users\asetyde\AppData\Local\V.Micro\Arduino\Builds\Garden\nodemcuv2\ESP8266WiFi\WiFiUdp.cpp.o"
binary.h:1:1: warning: null character(s) ignored [enabled by default]
binary.h:1:513: error: 'TNESS' does not name a type
binary.h:5:35: error: expected unqualified-id before 'if'
Error compiling

asetyde, 2015-08-17:
VisualStudio 2013 with visual micro , same in arduino 1.6.7 with staging esp library

igrr, 2015-08-17:
> binary.h:1:1: warning: null character(s) ignored [enabled by default]
binary.h:1:513: error: 'TNESS' does not name a type
binary.h:5:35: error: expected unqualified-id before 'if'

Something is wrong with your binary.h file.

asetyde, 2015-08-17:
How fix ?

asetyde, 2015-08-17:
with stable no error showed

asetyde, 2015-08-17:
ok report :

Compiling 'Garden' for 'NodeMCU 1.0 (ESP-12E Module)'
Build folder: file:///C:/Users/asetyde/AppData/Local/V.Micro/Arduino/Builds/Garden/nodemcuv2
Summary: Header=1 Prototypes=34 Imports=13
Additional Defines: 
Architecture Tools: C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/
Sketchbook: file:///C:/Users/asetyde/Documents/Arduino
Core Include Paths
Include Path 'C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266'
Include Path 'C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\nodemcu'
Library Include Paths (4)
Include Path 'C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\EEPROM'
Include Path 'C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\ESP8266WiFi\src'
Include Path 'C:\Program Files (x86)\Arduino\libraries\Rtc-master'
Include Path 'C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\Wire'
Platform Libraries Root Include Paths
Include Path 'C:\Program Files (x86)\Arduino\libraries'
Include Path 'C:\Users\asetyde\AppData\Roaming\arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries'
Include Path 'C:\Program Files (x86)\Visual Micro\Visual Micro for Arduino\Micro Platforms\default\debuggers'
Include Path 'C:\Users\asetyde\Documents\Arduino\libraries'
All import libraries will be re-compiled

asetyde, 2015-08-17:
No problem with stable , i'm waiting to staging be stable

closed 2015-08-17 by asetyde