esp8266/Arduino #703: Unaligned realloc for Strings

anmaped, 2015-08-18:
Webserver crashes when unaligned strings are passed as argument for write operations. Any kind of alignment is not provided.

Is os_realloc function supposed to return an aligned address such as found for other archs ?

igrr, 2015-08-18:
Could you attach an example stack trace?
os_realloc should return an aligned address due to the way heap is managed by ROM functions.
If it returns some unaligned address this may indicate that the heap was corrupted.

anmaped, 2015-08-18:
What I found sounds strange. I'm using the windows version of current development version. It appears that stack variables and heap variables may have an unaligned address but this does not happen for the static variables.

I'm dealing with that problem since some days ago, and in my application I didn't find anything that can be corrupting the memory. Stack size is more or less 250 bytes when exception (9) is triggered.

This only happen if I make successive calls to the webserver, 50 more or less. The tcp_out is the function that crashes with fatal exception (9).

anmaped, 2015-08-21:
I have a code that can duplicate the error of my first application.
The exception output is:
Fatal exception (9):
epc1=0x40101802, epc2=0x00000000, epc3=0x00000000, excvadd
Exception (9):
epc1=0x40101802 epc2=0x00000000 epc3=0x00000000 excvaddr=0x35a78a6f depc=0x00000000

ctx: cont 
sp: 3ffeb370 end: 3ffeb690 offset: 01a0

>>>stack>>>
3ffeb510:  40209781 3ffe9548 00000004 00000194  
3ffeb520:  3fff2d58 3ffeb570 3ffe9548 4020511c  
3ffeb530:  3fff3988 0000007c 0000007c 4020989a  
3ffeb540:  3ffe9d48 40241dc4 3ffeb570 3ffe957c  
3ffeb550:  3ffe9d48 00000000 3ffe9548 40205146  
3ffeb560:  00000000 0000000d 3ffeb5b0 4020236a  
3ffeb570:  3fff2870 00000004 00000004 3fff2d58  
3ffeb580:  00r=0x35a78a6f, depc=0x00000000
00000c 0000000c 3ffeb5b0 40209928  
3ffeb590:  40209781 3ffe9548 3fff2c68 40205782  
3ffeb5a0:  00000000 00000000 3ffe9548 4020536d  
3ffeb5b0:  3fff2d58 0000000d 0000000d 40203ae0  
3ffeb5c0:  3ffe9560 00000008 00000003 00000003  
3ffeb5d0:  3ffe88dc 00000002 000064a8 3fff2f80  
3ffeb5e0:  00000000 00000001 000003e8 3ffeb6bc  
3ffeb5f0:  00000000 3fff3050 3ffeb6f0 3ffeb6bc  
3ffeb600:  3fffdc20 000003e9 3ffe9548 40205461  
3ffeb610:  3ffe89f0 00000000 000003e8 00023498  
3ffeb620:  3ffe9560 3fff3050 3ffeb640 40208ed9  
3ffeb630:  3fffdc20 0000005e 3ffeb6f0 4020255e  
3ffeb640:  3fff2d78 00000012 00000012 3fff2870  
3ffeb650:  00000005 00000005 3fff8218 000022eb  
3ffeb660:  000022eb 00000000 4020237c 40202348  
3ffeb670:  3fffdc20 00000000 3ffeb6b4 402018ba  
3ffeb680:  00000000 00000000 3ffea670 40100374  
<<<stack<<<

If you try to refresh http://192.168.4.1/light several times 
```
#include <ESP8266WiFi.h> 
#include <ESP8266WebServer.h>

#define DEBUG

ESP8266WebServer server(80);

void handleRoot()
{
  
}

const char ss1[] PROGMEM = R"====(<!DOCTYPE html><html lang=\"en\"> <head>
    <title>ttt</title>
    <meta charset="utf-8" />
    <link rel="stylesheet" href="style.css"> 
    <script src="cscript.js"></script> 
    </head><body><img height="100px" src="logo.png"/>
  <nav class="dropdownmenu"><ul>
    <li><a href="info">Info</a></li>
  </ul>
</nav>123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789 
  0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901 
  7890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234 
  5678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345)====";
  
void handleLight()
{
  
  
  String x = "";
  x += ss1;
  uint8_t n = random(1,255);
  for (int i=0; i < n; i++)
  {
    uint8_t n2 = random(1,255);
    x += n2;
    x += FPSTR(ss1);
  }
  
  server.send ( 200, "text/html", x );
}

void handleAdvset()
{
  
}

void handleClock()
{
  
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  randomSeed(analogRead(0));

  WiFi.softAP("AAAA");

  server.on("/", &handleRoot);
  server.on("/light", &handleLight);
  server.on("/advset", &handleAdvset);
  server.on("/clock", &handleClock);
  
  server.onNotFound ( [=](){
    server.send ( 404, F("  text/plain"), F("    ") );
    } );
  
  // start server
  server.begin();
  
}

const char ss2[] PROGMEM = "123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789";
  
void loop() {
  // put your main code here, to run repeatedly:
  
  
  String x = "";

  uint8_t n = random(1,255);
  for (int i=0; i < n; i++)
  {
    uint8_t n2 = random(1,255);
    x += n2;
    x += FPSTR(ss2);
  }

  
  if (((unsigned int)x.c_str()) % 4 != 0)
     Serial.println(x);

 Serial.println("L#Heap Size: " + String(ESP.getFreeHeap()));

 server.handleClient();

}
```

igrr, 2015-08-31:
Ok, this is the same as #428 then.

igrr, 2015-09-30:
Fixed in staging.

closed 2015-09-30 by igrr