esp8266/Arduino #704: Add note that serial monitor doesn't work

ikeji, 2015-08-18:
I'm using Arduino IDE 1.6.5 which download from Arduino website and stable version of json file.
The board is ESP12 and connect like "Generic ESP8266 modules" in [boards page][1].
[1]: https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md#generic-esp8266-modules

This works with compile and upload program, but serial monitor menu doesn't works.
Because the RTS/DTR was enabled by serial monitor.

There is a patch disable RTS/DTR in serial monitor in esp8266/Arduino repository,
but I used Arduiono IDE which download from Arduino website and this enables RTS/DTR.

I feel this is very confusing.
Can you add some note who download ESP8266 package only?
Or can you prefer custom build Arduino IDE?

igrr, 2015-08-18:
Serial monitor does work, but reset circuit you use puts ESP into reset when you open the monitor. My suggestion is to use the circuit described in this comment instead: https://github.com/esp8266/Arduino/issues/480#issuecomment-122993574

ikeji, 2015-08-26:
Thanks for reply.

My ESP12 did different thing. It keep resetting until close the monitor.

When open the monitor, RTS/DTR was switch hi to lo and keep lo until I closed the monitor.
Some page said ESP8266 resets by down edge, but actually it doesn't run program until reset goes to high.
Doesn't this happen in your ESP?

Also I think we need staging version of json to use NodeMCU reset circuit.

ReMiA00, 2015-10-14:
I have tried several methods on resetting the ESP-12E for flashing with Arduino IDE
However none worked great, and all will break the "serial monitor" in the IDE

Since i had a ATtiny85 lying around i used this to send the reset to the ESP:
- The RTS pulse on my CP2102 lasts about 5 to 8 millis
- When serial monitor is started RTS remains low until the monitor is closed.
- To reset the ESP the RST pin needs to gow LOW while keeping GPIO0 LOW while booting the ESP

```
/*
  Reset ESP-12E with ATTiny85 (8Mhz)
 */

#include <SoftwareSerial.h>
#define rxPin 3  
#define txPin 4

SoftwareSerial mySerial(rxPin, txPin);

const int RTS_pin = 2; // Pin 7 on Attiny
const int RST_pin = 0; // Pin 5 on Attiny
const int GPIO0_pin = 1; // Pin 6 on Attiny

// Hookup 
// ESP     <--> Tiny pin <--> CP2102 or FT232RL
//  GND   <-->  4        <--> GND
//  5V    <-->  8 
//              7        <--> RTS 
//  RST    <--> 5            
//  GPIO0 <-->  6                  
//  RX <-------------------> TX
//  TX <-------------------> RX

unsigned long StartReset = 0;
unsigned long Puls = 0; 

void setup() {
  mySerial.begin(9600);

// USB to Serial Converter Flow Control RTS
  pinMode(RTS_pin,INPUT);

// ESP Pins
  pinMode(RST_pin,OUTPUT);
  pinMode(GPIO0_pin,OUTPUT);
  digitalWrite(RST_pin, HIGH);
  digitalWrite(GPIO0_pin, HIGH);
}

void loop() {
  int RTS = digitalRead(RTS_pin);
  
  if (RTS == 0 and StartReset == 0 ) { // Start Reset Puls
    StartReset = millis();
    mySerial.println("Reset");
  } 
  if (RTS == 1 and StartReset > 0 ) {  // Eind Reset Puls
    Puls = millis() - StartReset;
    mySerial.print("Puls = ");
    mySerial.println(Puls);
    StartReset = 0;
    
    if (Puls > 0 and Puls < 20) { // puls is around 5 to 8 millis
      digitalWrite(RST_pin, LOW);
      digitalWrite(GPIO0_pin, LOW);
      delay(5);        // length Reset puls
      digitalWrite(RST_pin, HIGH);
      delay(1000);        // Wait for ESP to boot
      digitalWrite(GPIO0_pin, HIGH);
      Puls = 0;
    }  
  }
}
```

I used a Arduino Nano with ArduinoISP sketch to program the ATTiny using IDE 1.6.5 
Add in "Boards Manager" https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json (see http://highlowtech.org/?p=1695)

Wire it up:
// Hookup to Program ATTiny85
// NANO   <--> Tiny pin
//  GND   <--> 4
//  5V    <--> 8
//  10    <--> 1
//  11    <--> 5
//  12    <--> 6
//  13    <--> 7

Choose ATTiny85 Board at 8Mhz and flash the Bootloader
Now you can upload the sketch to the ATTiny (use Arduino as ISP as programmer)

Wires for Serial debugging
// NANO   <--> Tiny pin
//  RX(0) <--> 3 serial monitor
//  TX(1) <--> 4 serial monitor

devyte, 2017-10-20:
Closing per #3655 and age.

closed 2017-10-20 by devyte