esp8266/Arduino #707: Cycle control probably fault

asetyde, 2015-08-19:
i ask an help because my control cycle not work ! or i'm sleeping or something not work on code / compiler or board firmware

[image]

i use 3 control cycle to manipulate hour from rtc (is correct) , to separate hour in 4 part  i add every 15 min , 24 value ,  in photo configuration :

		mind.scheduling[i][34][0][0] = 1;    // 10:30 -> 10+48 = 58 (10.30-10.45)
		mind.scheduling[i][58][1][0] = 1;    // 10:45 -> 10+72 = 82 (10:45-11.00)
		mind.scheduling[i][11][2][0] = 1;    // 11.00 -> 11+0 = 11 (11.00-11.15)
		mind.scheduling[i][11][2][0] = 1;    // 11:15 -> 11+24 = 35 (11:15-11:30)
		mind.scheduling[i][35][3][0] = 1;    // 11:30 -> 11+48 = 59 (11:30-10.45)

asetyde, 2015-08-19:
it's strange that at 11.05  -> H = 11  not 35 !

asetyde, 2015-08-19:
now work but ... < c > ???

[image]

igrr, 2015-08-19:
`if (15 > c < 30)` looks funny, are you sure you this is what you intend to do?
`15 > c` will return true or false, depending on the value of c. This bool value will be converted into an integer (1 or 0). This 0 or 1 in turn will be compared with 30.

What you probably want is `if (15 > c && c < 30)`.

One more thing: please use esp8266.com forum for coding help. This issue tracker is not really intended for questions.

asetyde, 2015-08-19:
it's not unique expression as in math ?

asetyde, 2015-08-19:
ok thx , i think it's my code error but i want to see if something wrong .

asetyde, 2015-08-19:
another out quest , because i not understand how can i write to you , i use arduino standard with loaded library , for unaware i must delete and put this in your repository ? 

thx @igrr

igrr, 2015-08-19:
You might want to check C++ language reference regarding comparison operators: http://en.cppreference.com/w/cpp/language/operator_comparison

Another option is to use GCC [case ranges](https://gcc.gnu.org/onlinedocs/gcc/Case-Ranges.html) extension:
```c++
  switch (value) {
    case 0 ... 32:
      Serial.println("this is small");
      break;
    case 33 ... 64:
      Serial.println("this is just fine");
      break;
    case 65 ... 255:
      Serial.println("probably too large");
      break;
  }
```

closed 2015-08-19 by igrr