esp8266/Arduino #710: Unable to upload Sketch

stblassitude, 2015-08-19:
I'm having trouble uploading the blink sketch to my ESP-01. I'm on MacOS 10.10.5, with Arduino 1.6.5 and the stable board manager (http://arduino.esp8266.com/stable/packa ... index.json). I've also tried the staging version, but no difference.

When I upload the sketch, I get this log, and the sketch does not appear to be working. Using themadinventor's esptool.py, I am able to upload the nodecmu image to the very same module. I'm using an PL2303 cable (Adafruit clone).

I've also played around with the various settings (flash mode, speed, etc.), but to no avail.

Any suggestions?

```
Der Sketch verwendet 201.090 Bytes (46%) des Programmspeicherplatzes. Das Maximum sind 434.160 Bytes.
Globale Variablen verwenden 44.608 Bytes (54%) des dynamischen Speichers, 37.312 Bytes für lokale Variablen verbleiben. Das Maximum sind 81.920 Bytes.
/Users/stb/Library/Arduino15/packages/esp8266/tools/esptool/0.4.5/esptool -vv -cd ck -cb 115200 -cp /dev/cu.usbserial -ca 0x00000 -cf /var/folders/r9/9t0hn90919jdxnztplqfvcxw0000gp/T/build3191704045558840700.tmp/Blink.cpp.bin 
esptool v0.4.5 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
   setting board to ck
   setting baudrate from 115200 to 115200
   setting port from /dev/tty.usbserial to /dev/cu.usbserial
   setting address from 0x00000000 to 0x00000000
   espcomm_upload_file
   stat /var/folders/r9/9t0hn90919jdxnztplqfvcxw0000gp/T/build3191704045558840700.tmp/Blink.cpp.bin success
opening port /dev/cu.usbserial at 115200
   tcgetattr
   tcsetattr
   serial open
opening bootloader
resetting board
trying to connect
   setting character timeout 0
   done
   setting character timeout 1
   done
   espcomm_send_command: sending command header
   espcomm_send_command: sending command payload
trying to connect
   setting character timeout 0
   done
   setting character timeout 1
   done
   espcomm_send_command: sending command header
   espcomm_send_command: sending command payload
   espcomm_send_command: receiving 2 bytes of data
   espcomm_send_command: receiving 2 bytes of data
   espcomm_send_command: receiving 2 bytes of data
   espcomm_send_command: receiving 2 bytes of data
   espcomm_send_command: receiving 2 bytes of data
   espcomm_send_command: receiving 2 bytes of data
   espcomm_send_command: receiving 2 bytes of data
   espcomm_send_command: receiving 2 bytes of data
   espcomm_open
Uploading 205232 bytes from /var/folders/r9/9t0hn90919jdxnztplqfvcxw0000gp/T/build3191704045558840700.tmp/Blink.cpp.bin to flash at 0x00000000
   erasing flash
   size: 0321b0 address: 000000
   first_sector_index: 0
   total_sector_count: 51
   head_sector_count: 16
   adjusted_sector_count: 35
   adjusted_size: 023000
   espcomm_send_command: sending command header
   espcomm_send_command: sending command payload
   setting timeout 10000
   setting character timeout 100
   done
   setting timeout 1
   setting character timeout 1
   done
   espcomm_send_command: receiving 14850 bytes of data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader```

stblassitude, 2015-08-19:
See also http://www.esp8266.com/viewtopic.php?f=26&t=4597

igrr, 2015-08-19:
This is a known issue with PL2303: https://github.com/igrr/esptool-ck/issues/9
I can reproduce it myself but I can't figure out what is going on. See my comments in the linked issue.
Let's move further discussion on to the esptool-ck issue tracker.

closed 2015-08-19 by igrr

jyotirajsharma, 2016-12-27:
Hi,

I am getting following errors while uploading my code to arduino.
I am trying to use esp8266 chip and correctly connected all the pins. I can see esp8266 chip is getting powered on properly. But while loading the code, following errors occurs. I tried searching online solutions but there is no luck. Please help me to resolve this issue as soon as possible.

Sketch uses 236,389 bytes (54%) of program storage space. Maximum is 434,160 bytes.
Global variables use 32,568 bytes (39%) of dynamic memory, leaving 49,352 bytes for local variables. Maximum is 81,920 bytes.
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

vndkumar, 2017-03-07:
Connect 
Tx (ESP) - Tx (ARDIUNO or USB_2_TTL )
Rx(ESP) - Rx (ARDIUNO or USB_2_TTL )
Vcc, CH_PD - 3.3v
For programing: GPIO0- GND, GPIO2-VCC
For Booting from Flash: rst once and GPIO0-VCC, GPIO2-VCC
ALSO * rst of (ARDIUNO or USB_2_TTL ) should be connected to GND.
SET the baud rate correctly, in my case it was 115200
Seletc bord : Generic ESP8266
IT will work fine.

eilering, 2022-12-28:
In the circuit the GPIO0 pin is connected to VCC through a 10 k resistor. The voltage on the reset pin is 1.97V. in normal operation.
I removed this resister and connected the GPIO0 pin manually.
Normal operation--> GPIO0 to VCC
Program mode --> GPIO0 to GND--> switch on VCC --> GPIO0 to VCC.

I tried it with 5 esp8266 modules which all were functioning properly after this modification and did not before.