esp8266/Arduino #712: ESP and DMX Library

folny, 2015-08-19:
Hi

Someone worked with libraries to transfer DMX with Arduino ESP8266 example Conceptinetics, DMXSerial or DmxSimple or know someone how I could edit this library to run under the ESP https://github.com/mathertel/DmxSerial ?.

mc-hamster, 2015-08-19:
DMX512 is just Asynchronous serial at 250,000 baud, 8 bits, no parity and 2 stop bits. 8n1 also works on the receive (don't try it on a sender). There's a short pause between packets.

You can use any serial library capable of those specs.

- Jm

folny, 2015-08-19:
Thanks for responding

Can you tell me specifically which library I could use or a project that uses such a library ?.

mc-hamster, 2015-08-19:
The esp8266 serial library can do this. Check the arduino website for docs on how it works.

- Jm

folny, 2015-08-20:
Hi

In my case I need a DMX signal to send, here is the code which I tried but it did not work properly like I could do to make it work properly ?.

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

char ssid[] = "SSID";  
char pass[] = "PASS"; 
WiFiUDP udp;
unsigned int localPort = 5568;  

#define E131_BUFFER_SIZE 638 

unsigned char E131Buffer[E131_BUFFER_SIZE]; 

void setup()
{
  Serial.begin(250000,SERIAL_8N1); //500,000 bits per second to keep up with UDP data coming in 
  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(500);
  }
  udp.begin(localPort);
}

//checks to see if packet is E1.31 data
int checkE131Headers(unsigned char* messagein, int messagelength) 
{
  if ( messagein[1] == 0x10 && messagein[4] == 0x41 && messagein[12] == 0x37) 
  {  
      int ChannelCount = messagein[123] * 256 + messagein[124]; // number of values plus start code
      return ChannelCount -1; //Return how many values are in the packet.
    }
  return 0;
}

void loop() 
{
  int packetSize = udp.parsePacket(); //parse packet
  if(packetSize)
  {
    udp.read(E131Buffer,E131_BUFFER_SIZE); //read packet into buffer
    
    int count = checkE131Headers(E131Buffer,E131_BUFFER_SIZE); //pass data to see if it is E1.31
    if ( count ) //if it comes back genuine
    {
      for(int i = 4; i < E131_BUFFER_SIZE; i++) 
     {
      Serial.write(E131Buffer[i]); //write bytes 4-638 in packet to serial port.  
     }
     Serial.flush(); //flush transmit buffer at end of for loop
    }
  }
} //end loop

Rickgg, 2015-11-03:
Hi, I am currently working on a [DMX library](https://github.com/Rickgg/ESP-Dmx). I haven't been able to test it because I lack the FTDI programmer, but by logic it should work.
If you test it I would be very happy if you posted results :smile:

bfranske, 2016-01-05:
FYI I have run some basic tests on Rickgg's DMX library (which he linked to above) using a NodeMCU ESP8266-12E based board and a TI SN75176BP RS485 driver chip. The library seems to work for DMX transmission from the ESP8266 using UART1, only tested with three DMX channels for right now.

rgr101, 2016-02-16:
Anyone who ever tried to run DMXSimple on the ESP8266?
Afaik, this lib uses just one digital i/o in conjunction with a very simply attached SN75176 
and so communication may be easier than going through the serial interface?

bfranske, 2016-02-16:
FYI There's nothing magical about the SN75176 that makes communication easier, it's a regular serial to RS-485 transceiver. If someone is using it with a regular digital I/O that's because they're using software serial instead of hardware serial, you could do the same thing with other DMX libraries as well. A disadvantage though is that DMX is fairly timing sensitive and you're likely to get better timing from  a real UART interface than from bit-banging on a digital line. As far as using just one pin, that's simply a case of creating a transmit only interface, again other libraries can do the same thing if you're only transmitting DMX and don't need to receive DMX, the chip is just hardwired into always transmit mode.

forkineye, 2016-07-18:
The only tricky part is getting the DMX BREAK and MAB.  The ESP has native break support though so it makes it easy.  Just do the following before starting your 250k 8n2 stream:
```
/* UART for DMX output */
#define SEROUT_UART 1

/* DMX minimum timings per E1.11 */
#define DMX_BREAK 92
#define DMX_MAB 12

...
SET_PERI_REG_MASK(UART_CONF0(SEROUT_UART), UART_TXD_BRK);
delayMicroseconds(DMX_BREAK);
CLEAR_PERI_REG_MASK(UART_CONF0(SEROUT_UART), UART_TXD_BRK);
delayMicroseconds(DMX_MAB);
...
```
Full implementation here - https://github.com/forkineye/ESPixelStick.  You'll want to look at SerialDriver.*

devyte, 2017-10-20:
Closing per #3655 .

closed 2017-10-20 by devyte