esp8266/Arduino #715: Interrupts on high GPIO

Toshik, 2015-08-21:
It seems, that interrupts attached to high GPIOs does not works. 

I have tried to attach interrupt to GPIO14 - it do not reacts to anything. Then I have tried to change it to GPIO2 - and it works fine. 

To be sure - GPIO14 works fine as regular output.

anwarhahjjeffersongeorge, 2015-09-03:
Does the pinMode make a difference?

Toshik, 2015-09-03:
@anwarhahjjeffersongeorge I have tried INPUT/OUTPUT - all same

anwarhahjjeffersongeorge, 2015-09-03:
I am not sure what your code is like, can you try this sketch on your ESP8266. Connect a jumper from GPIO14 to ground (or whatever pin you want to test) and move it back and forth between ground and high rails to see onboard LED switch state.

```
//connect a jumper from GPIO 14 to ground to start test

#define GPIO_PIN 14 

uint8_t led = LOW;
 

void setup(){
	pinMode(GPIO_PIN, INPUT);
	pinMode(BUILTIN_LED, OUTPUT);
	setLED();
	attachInterrupt(GPIO_PIN, highInterrupt, RISING);

}

void highInterrupt(){
	led = HIGH;
	detachInterrupt(GPIO_PIN);
	attachInterrupt(GPIO_PIN, lowInterrupt, FALLING);
}

void lowInterrupt(){
	led = LOW;
	detachInterrupt(GPIO_PIN);
	attachInterrupt(GPIO_PIN,  highInterrupt, RISING);
}

void setLED(){
	digitalWrite(BUILTIN_LED, led);

}

void loop(){
	setLED();    
}
```

The above seems to work for me. I just compiled my Arduino environment from the repo a few days ago. Are you working with an older version?

igrr, 2015-09-08:
I have tried `CHANGE` interrupts on GPIO13 and they seem to work as well with the following sketch:
```c++

const int pin = 13;

void toggle() {
  static int state = 0;
  state = !state;
  digitalWrite(BUILTIN_LED, state);
}

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);
  pinMode(pin, INPUT);
  attachInterrupt(pin, toggle, CHANGE);
  toggle();
}

void loop() {}
```

marcellocampione, 2016-03-07:
Never execute a delay() in an interrupt procedure especially if there is the danger that you run alyo a delay in the loop().
Leads to immediate reset !

Toshik, 2016-03-07:
@marcellocampione where do you see delays here?

marcellocampione, 2016-03-17:
Just a general reccomendation. I used a small delay for debouncing an interrupt driven button... and was exceriencing only resets. 
Now it works fine with something like:

//****************************************
// HandlePlus
//****************************************

void HandlePlus() {
  int Milliseconds;
 
  detachInterrupt(PlusPin);

  // Increment the Temperature value f.i.
  if (Temperature<25) { Temperature+=0.5; }

  //debounce delay
  Milliseconds = millis()+50;
  while (Milliseconds > millis()) ;

  Serial.print("Manual Temperature Change: ");
  Serial.print(Temperature);
 //Other debounce and repeat trick
  while (digitalRead(PlusPin) == 0) { 
        //Serial.println(Temperature);
  }

pieman64, 2016-04-11:
When I try to use attachInterrupt with GPIO 0, 2, 12 and 15 (spare pins I currently have available) I see  the high and low in Serial Monitor when I short to ground and then it almost always resets the ESP. I'm using a breakout board for the ESP07 with regular pullup resistors used for flash mode.

Is there something else I should be doing or adding to stop the ESP resetting when I ground the pins?

pieman64, 2016-04-11:
Ok with GPIO 12 and changing pinMode from INPUT_PULLUP to just INPUT the resets have stopped. Using attachInterrupt in CHANGE mode I can 'toggle' by shorting to ground and then Vcc alternately but I just want the momentary switch tied to ground (not ground and Vcc).

How would I do this and how many pin interrupts can the ESP handle?

pieman64, 2016-04-12:
I finally managed to get ESP to work with attachInterrupt but I believe it can't do WiFi AND interrupts at the same time. So it has to be one or the other.

Vargha, 2016-08-19:
Hi guys, I explained it on my website, please check it out if you like:
http://subtledesign.net/index.php?id=learning-lounge&post=esp8266-setting-interrupts

luffykesh, 2016-11-15:
For me, Interrupt does not work while I am constantly trying to connect to WiFi
while(WiFi.status()!=WL_CONNECTED)
{
 Serial.print(".");
delay(1000);
}
I have declared my interrupt as: 
attachInterrupt(digitalPinToInterrupt(BUTTON), buttonISR, CHANGE);

Is there an issue here in my code, or is there an another issue?

Vargha, 2016-11-15:
Hi luffykesh,
I explained it in a video, and I hope it would be helpful to you:
[https://www.youtube.com/watch?v=JqF6677ijLU](ESP8266 - Setting Interrupts for Pins )
If you can use your interrupt pin as output, easy. otherwise, you need to check your incoming voltage to the interrupt pin, and you cannot drop the voltage using a voltage regulator, if it is high. Any voltage higher than 3.3V needs to drop to 3.3V using resistors (something real-time). If still didn't work, leave me a comment under the video, but honestly that's all what I can say.

luffykesh, 2016-11-15:
Hello Vargha, 
The pin on which interrupt is set is a pushbutton switch, and on pressing I have to perform some task, and while ESP is connecting to a station, it does not respond to the button press. Once connected, the interrupt works fine.

Vargha, 2016-11-15:
Try this:
Change the GPIO pin to output, and I think it can work in your situation.
Change the interrupt mode to raising or falling (not change), and on your setup set the pin to low (if rising), or high (if falling)
Remove the button, and manually connect it to VCC, and then to ground, see if it functions (I hope it will)
Then, you'll at least have an idea if your interrupt is working at all. I'm putting my code example here:

void interruptFunction()
{
  LCDClear();
  gotoXY(0, 3);                              // Going to the line 3 on my LCD
  LCDString("Interrupted");            //   Print something on the LCD
  digitalWrite(13, 0);                    //     Setting GPIO back to 0, to be ready for future interrupts
}

/* -------------------------------------------- SETUP --------------------------------------------*/
void setup() {
  pinMode(13, OUTPUT);                    // Next line won't work if the pin is not output
  digitalWrite(13, 0);                            // It should be LOW to be able to see the rise
  attachInterrupt(13, interruptFunction, RISING);          // set pin 13 to jump to the required function as soon as it connects to 3.3V

ccoenen, 2017-07-04:
I observe something very similar. I have working code with an `attachInterrupt(2, someISRFunction, CHANGE);`

Pin 2 works, but I need Pin 2 for something else now. I switched this to be pin 13 or 14, all of which don't work at all. I tried this with various pinModes and with a 10K Pull-Up resistor, nothing works.

Did anyone come up with a solution?

ccoenen, 2017-07-04:
I DID get this to work, apparently, It's on Pin 12, which is not my favorite, but it seems to be alright:

```c
const uint8_t BUTTON_PIN = 12;
const uint8_t LED_PIN = 0;

void buttonChange() {
  if (digitalRead(BUTTON_PIN) == LOW) {
    digitalWrite(LED_PIN, LOW);
  } else {
    digitalWrite(LED_PIN, HIGH);
  }
}

void setup() {
  pinMode(BUTTON_PIN, INPUT_PULLUP);
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);
  attachInterrupt(digitalPinToInterrupt(BUTTON_PIN), buttonChange, CHANGE);
}

void loop() {
  // nothing, you can to whatever you like, here.
}
```

This has a regular switch connected to Pin 12 and Ground, no other parts. Pin 0 is a LED which turns on when `LED_PIN` goes `LOW` (this is an onboard LED on my particular board).

devyte, 2017-10-20:
@Toshik no board specified, no MCVE sketch.
Is this issue still valid with latest git?