esp8266/Arduino #716: digitalWrite writes after yield() or delay(0)

Duality4Y, 2015-08-22:
I am having a peculiar issue, 
I am trying to write a pin low and high. and notice that it doesn't take effect immediately,
my application requires it to be immediate, because it triggers a conversion (ADC) and has to be read directly after. I had scoped the output, to find the problem.
[code]
uint16_t readAnalogChannel(int channel)
{
    int adcvalue = 0;
  int b1 = 0, b2 = 0;
  int sign = 0;

  // command bits for MCP3304
  // 0000 = diff, ch0 = in+, ch1 = in-
  // 0010 = diff, ch2 = in+, ch3 = in-
  // 0100 = diff, ch4 = in+, ch5 = in-

  digitalWrite (SELPIN, LOW); // Select adc
  delay(0);

  // first byte
  // first byte will always be B000010xx where xx are the D2 and D1 channel bits  
  byte commandbits = B0000110;
  commandbits |= (channel >> 2);         // high bit of channel

  SPI.transfer(commandbits);       // send out first byte of command bits

  // second byte; Bx0000000; leftmost bit is D0 channel bit
  commandbits = B00000000;
  commandbits |= (channel << 6);        // if D0 is set it will be the leftmost bit now
  b1 = SPI.transfer(commandbits);       // send out second byte of command bits

  // hi byte will have XX0SBBBB
  // set the top 3 don't care bits so it will format nicely
  b1 |= B11100000;
  //Serial.print(b1, BIN); Serial.print(" ");
  sign = b1 & B00010000;
  int hi = b1 & B00001111;

  // read low byte
  b2 = SPI.transfer(b2);              // don't care what we send
  //Serial.print(b2, BIN); Serial.print("\r\n");
  int lo = b2;
  digitalWrite(SELPIN, HIGH); // turn off device
  delay(0);

  int reading = hi * 256 + lo;

  if (sign) {
    reading = (4096 - reading) * -1;
  }

  return (reading);
}
[/code]

Links2004, 2015-08-22:
the maximum GPIO change frequency is limited by espressif to ~8Mhz.

but you can speed up you SPI code by using ```SPI.transferBytes```
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/SPI/SPI.cpp#L403

the fastest possible way to change the GPIOs is using the ```GPOC``` and ```GPOS``` register.
see here for example:
https://github.com/esp8266/Arduino/blob/esp8266/libraries/Adafruit_ILI9341/Adafruit_ILI9341.cpp#L159

Duality4Y, 2015-08-23:
I see And I understand I think, but spi clock freq is set to 10MHz.
and the point I am trying to make is that when i do a digitalWrite(pin, low) it doesn't do so "immediately",
what i meen is that the code under the digitalWrite runs before the pin is pulled low. and that is not allowed.

sticilface, 2015-08-23:
This might be related to the issue that Makuna had with the WS2812s.  The first bit was always longer for some reason.  He solved it in this commit

https://github.com/Makuna/NeoPixelBus/commit/859650448806a47357ed9f6ffd7e2ee4f17d60e8

by using ICACHE_RAM_ATTR

and moving the function to a C file, as the arduino IDE seemed to ignore it if it was in a cpp file.  I could most certainly be wrong though.

devyte, 2017-10-20:
I think the previous comment about ICACHE_RAM_ATTR is relevant. Without it, there could be a flash access triggered, causing odd behavior and crashes. Also, for timing-sensitive code, there are other things to consider (e.g.: code for reading DHT sensors).
Not the right place to discuss this. Closing per #3655 .

closed 2017-10-20 by devyte