esp8266/Arduino #717: nodemcu dev kit V1.0 (aliexpress V2) warning: espcomm_sync failed error: espcomm_open failed

prussiap, 2015-08-23:
I was able to flash this board but it felt random (aka when i sometimes unplugged or pushed flash button for a few seconds.

I'm on the newest OSX version all updates. Also I've tried switching from Stable to Staging and back (removing rebooting and changing the .json link)

I have Board selected as NodeMcu 1.0
speed 115200 though i've tried down to 9600
port is the correct one. 

My friend on linux doesn't need to push flash buttons or anything

I stumbled on this post: http://www.esp8266.com/viewtopic.php?p=25718 but not sure if any fixes since.

mc-hamster, 2015-08-23:
I ran into something very similar to one of the boards I got from Aliexpress. Good to know that my experience was not unique. Noticed that the USB bridge chip got unusually warm and the bench power supply was providing more power than it should have for the situation. I didn't perform any further trouble shooting. It's now in the trash bin.

- Jm

liquiddandruff, 2015-08-25:
Can we see the product listing?

prussiap, 2015-08-25:
sure: http://www.aliexpress.com/item/V2-4M-4FLASH-NodeMcu-Lua-WIFI-Networking-development-board-Based-ESP8266/32264797795.html  they name version differently but pretty sure its 1.0

igrr, 2015-08-25:
Using this one with OS X, works quite well, I can upload at 921600. Check
that you have Silabs driver installed, try changing the USB cable. Try
uploading something with NodeMCU flasher utility.

prussiap, 2015-08-25:
Yep i have silabs driver and other two popular ones, had tried cables, rebooting etc before posting as well. Also this works fine without flash buttons etc with these variants. http://www.aliexpress.com/item/V2-4M-4FLASH-NodeMcu-Lua-WIFI-Networking-development-board-Based-ESP8266/32388289174.html

not sure what else to do. i'll test nodemcu variant when i can.

luciancerbu87, 2015-08-26:
I have this board as well, initially I only uploaded at 57600, but I start playing around with different values and get to 262k, but how @igrr says I think I will try the upper limit.
The issue I had with this board was entering deep sleep mode with low consumtion < 0.3mA (you can see https://github.com/esp8266/Arduino/issues/719 if you are interested for a wireless sensor node)

asetyde, 2015-08-26:
you check DIO / QIO on settings ? my nodemcu work good , your esp is 12E not 12

prussiap, 2015-08-26:
@luciancerbu87 I've tried a variety of speeds from 9600 to 57600 and I would love to have the deep sleep issue but my problem is just uploading a sketch. Way before anything useful happens. That and the Lolin boards are the ones that work just fine.. It's too bad though that they barely fit a breadboard :). 

@asetyde what do you mean by check DIO/ QIO? I don't have either of those settings available since I used the NodeMCU V1.0 (esp12-e)

I have tried the generic stuff just to see if that could work but don't remember what I tried and it never worked. 

These boards do seem to work without hitting flash button for my friend on linux so whatever the problem is seems to be osx, drivers, arduino ide issue on my part. 

I've uninstalled and re-installed all the FTDI chip drivers (ch34 etc.. ) removed the arduino esp .json drivers and stuff quit reboot restarted. etc.. 

I'm at a bit of a loss and not sure if I can provide any log info that can help.

Also I've added some debug lines for the Lolin one to test wifi connection etc.  Serial.begin(9600); and   Serial.println(F("Booting...."));

This should be working just fine from Arduino IDE Serial Monitor (set at 9600) with just the usb cable plugged in. I don't need to switch to using FTDI tx/rx right ? Pretty sure it worked just fine before using usb cable but just checking.

mc-hamster, 2015-08-26:
@prussiap - That's the same product listing that I purchased from and thew out.

prussiap, 2015-08-26:
@mc-hamster which devkit or lolin ? my friend had no issues with it on his linux box so not sure... other people on forum said they worked.

mc-hamster, 2015-08-26:
@prussiap http://www.aliexpress.com/item/V2-4M-4FLASH-NodeMcu-Lua-WIFI-Networking-development-board-Based-ESP8266/32388289174.html ... I purchased a lot of 5 and one of them had issues.

prussiap, 2015-08-26:
interesting. Well i'm pretty sure I fried one (magic smoke and smell) but other one was flashed once it seemed and then that was it.

asetyde, 2015-12-01:
no way to maintain this , please close issue

closed 2015-12-01 by Links2004