esp8266/Arduino #719: Can't activate deep sleep mode, all the time current consumption is 12mA

luciancerbu87, 2015-08-24:
Hello, 

I have bought this ESP8266 : 
[image]
with the following piout : 
[image]
this is the current link : http://www.aliexpress.com/item/2PCS-V3-4M-bytes-32Mbits-FLASH-NodeMcu-Lua-WIFI-Networking-development-board-Based-ESP8266-with-firmware/32368521069.html

So the problem is that I can't set it to deep sleep, the minimum consumption that I can obtain is 12mA on 3.8V. I made a link from DO(GPIO16) to RST and after deep sleep it restarts itself , but I can't get it to uA consumption, how it described in documentation.

Have you got any ideea, why I have this problem? 

Thanks is advance.

Links2004, 2015-08-24:
this board has an USB to uart chip, this chip also consumes energy.
where exactly do you measure the power consumption?

CH430:
[image]

luciancerbu87, 2015-08-24:
I think that is the problem,  I should think of a method to take that out and measure again.  Thank you lot!

luciancerbu87, 2015-08-24:
Ok, so after trying to unsolder VCC from CH430G, I get this : 
[image]
the problem is that... after disconnecting the UART I get a consumption of 10mA and if I connect it again I get  10.1 mA... 
The problem before was that I was supplying voltage to 3V pin and ground, now I'm supplying from VIN pin and GND

This is my code : 

/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */
//deep sleep include
extern "C" {
#include "ets_sys.h"
#include "os_type.h"
#include "osapi.h"
#include "mem.h"
#include "user_interface.h"
#include "cont.h"
}
const uint sleep_timer_user = 1; // sleep in minutes 
ADC_MODE(ADC_VCC);
#include <ESP8266WiFi.h>
#include <OneWire.h>
#include <DallasTemperature.h>
#include <stdlib_noniso.h>
 
 
#define ONE_WIRE_BUS D1  // DS18B20 pin
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature DS18B20(&oneWire);
 
const char* ssid     = "Oglinzilor_2.4Ghz";
const char* password = "apartament1";
 
const char* host = "api.thingspeak.com";
const char* APIkey   = "V0RUY16H7C0327WE";
 
 
float oldTemp;
 
 
 
void setup() {
   //system_deep_sleep_set_option(4);
  Serial.begin(9600);
  delay(10);
 
  // We start by connecting to a WiFi network
 
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
 
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}
 
 
 
void loop() {

 
  float temp;
 
 
  DS18B20.requestTemperatures();
  temp = DS18B20.getTempCByIndex(0);
  Serial.print("Temperature: ");
  Serial.println(temp);
 
  char charVal[12];
 
  dtostrf(temp, 8, 2, charVal);
 
  Serial.print("connecting to ");
  Serial.println(host);
 
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
 
 
 
 
  // We now create a URI for the request
  String url = "/update?key=";
  url += APIkey;
  url += "&field1=";
  url += charVal;//String(temp);
 
 
 
  Serial.print("Requesting URL: ");
  Serial.println(url);
 
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Connection: close\r\n\r\n");
  delay(10);
 
  // Read all the lines of the reply from server and print them to Serial
  while (client.available()) {
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
 
  Serial.println(String("Voltage = ")+ ESP.getVcc() +String(" V"));
  Serial.println("closing connection");
  //WiFi.mode(WIFI_OFF);
   //delay(900000);
//wifi_set_sleep_type(NONE_SLEEP_T); 
//system_deep_sleep_set_option(4);
//Serial.print("wifi Sleep type = ");
//Serial.println(wifi_get_sleep_type());

//ESP.deepSleep(sleep_timer_user*60*1000*1000);
 // ESP.deepSleep(sleep_timer_user*60*1000*1000,WAKE_RF_DEFAULT); //19mA with step-up
 // ESP.deepSleep(sleep_timer_user*60*1000*1000,WAKE_RFCAL);//19mA with step-up
   //ESP.deepSleep(sleep_timer_user*60*1000*1000,WAKE_NO_RFCAL);//19mA with step-up
   ESP.deepSleep(sleep_timer_user*60*1000*1000,WAKE_RF_DISABLED);//19mA with step-up
}

luciancerbu87, 2015-08-24:
So after some measurements I found this consumption on different deep sleep modes : 
//ESP.deepSleep(sleep_time);//12mA without UART
//ESP.deepSleep(sleep_time,WAKE_RF_DEFAULT); //31mA without UART
//ESP.deepSleep(sleep_time,WAKE_RFCAL);//12.4 mA without UART
//ESP.deepSleep(sleep_time,WAKE_NO_RFCAL);//10-10.3mA working wifi after reset without UART
//ESP.deepSleep(sleep_time,WAKE_RF_DISABLED);//10-10.3mA working wifi after reset without UART

I couldn't activate deep sleep for a consumption of uA... maybe this board have some regulators that consume 10mA, but I don't know. If anyone can give me a tip.

Thanks.

Links2004, 2015-08-24:
yes  biggest 3 PIN component looks like a regulator, shut make 5V to 3V3.
the board looks more or less like the nodemcu excepted the power supply.

[image]

i only use the ESP12 and ESP7 boards directly with my custom boards. never had a problem with it.
have you a schematic for you board?

you can try to unsolder the 3V3 pin of the LDO (regulator) and attache the ampere-meter there.

duncan-a, 2015-08-24:
Does this board have LEDs? They're a power drain...

I agree with what Markus is saying - you won't minimise the power
consumption with a load of bells and whistles on the board.

luciancerbu87, 2015-08-24:
Well thanks for the support, after a lot of soldering and re-soldering and desoldering a lot of pins :) I finally get where the problem was.
1)the problem is not the UART chip, it has some sleep mode which enters.
2)the regulator is the problem it uses power.

To make it simple, all you need to do for deep sleep is to cut PIN 1, and supply directly the ESP12-E.

[image]
In my case, I used o wire for the pin1 and a wire to the solder pad of pin1, in this case when I need to program it over USB where I have 5V, I just connect the wires and it regulates the voltage to 3.3, but when it runs stand-alone, I disconnect and it enters sleep mode.

[image]
[image]
[image]

So the final consumption is 250-270uA in deep sleep, I will try to cut down the UART chip in order to have a lower consumption, but today it is enough.

Again, thank you @Links2004 for interest.

closed 2015-08-24 by luciancerbu87

dmody, 2016-04-20:
Hi!  
I took a stab at reproducing your work and wasn't able to get the current value below 7.8 mA.   

The code I'm using is as follows:
#include <ESP8266WiFi.h>

int pin=2;//wifi LED   since you can't use the builtin LED because it's being used for wakeup (it's D0 = 16)

void setup() {
  WiFi.mode( WIFI_STA );
  pinMode(pin, OUTPUT);     // Initialize the  pin as an output

}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(pin, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(3000);                      // Wait for a second
  digitalWrite(pin, HIGH);  // Turn the LED off by making the voltage HIGH

  //put into deep sleep for awhile

ESP.deepSleep(9000000, WAKE_RF_DISABLED); // Sleep for 9 secs

dmody, 2016-04-20:
I unsoldered Pin 1 from the voltage regulator from the circuit board and added two jumpers.   One jumper from the pad of the voltage regulator to the ground on a 3.7V battery.  The other jumper on Pin 1 of the voltage regulator is hanging loose.  I then added a jumper from the ESP8266 module per your photograph the positive side of the battery.  The code runs well and the current is as follows:

80mA on startup (that's the wifi module pulling that).   The wifi module blue light illuminates due to me pulling pin 2 low.

It appears to enter deeps sleep and the current is 7.8 mA.

9 seconds later it comes out of deep sleep and pulls 25 mA.  There is a flicker in the onboard LED (due to wakeup) and the wifi led glows solid and bright.  

The unit then goes into sleep and the current is 7.8 mA.  The last two steps repeat.

I can't seem to locate the source of the 7.8 mA drain.  I have tried unsoldering the regulator completely and that does nothing.

vlast3k, 2016-04-20:
which board are you using? If it is one equipped with a usb2serial chip
that is powered from the 3v, e.g. nodemcu v1 using CP2102, then this is the
source
also try using 2xaa batteries
a 3.7v lipo battery gives 4+v when charged. and if you are bypassing the
voltage regulator the module gets lots of voltage

luciancerbu87, 2016-04-20:
@dmody : well there is the usb-to-serial converter, you must cut it's power too. I will try to do it myself the next week, and put out a clear tutorial.

@vlast3k : after bypassing the regulator and cut power of usb-to-serial (CH430G in my case), I get from ADC port around 3.6-3.7V, when battery is fully charged 4.2V. 
It worked for me this way for about, 8months. I know it's not ok, because from 3.3V to 4.2V it's a long way, but for me it works.

vlast3k, 2016-04-20:
In fact do you need to cut the voltage regulator? What i did was to supply the power directly via the 3.3 v input.
But on the nodemcu v1 i was still getting around 15ma, and this is why i suspected the CP2101
so i switched to the nodemcu v0.9 with the CH340g, which is powered from the 5v line, and then i got 90 ua
i supposed in your case the 240 ua, are due to you are powering it with higher voltage and it consumes a bit more, but if worked for so long time - apparently it is not that weak :)

duncan-a, 2016-04-20:
Basically, I think you'd be better to use a different module - what you're
trying to do is start from a very complete module and strip away everything
that's using power, why not start with a basic module that doesn't have any
of the extras?

I'm running an ESP-12, with the LEDs removed, in deep sleep mode, waking
every ten minutes, and the current is too low for my multimeter to read
when it's sleeping. There's nothing but the basics so it should be around
the quoted 7.8uA (if that includes the Flash chip).

Start with the absolute minimum and add the parts you need - thay way you
know you're not powering anything you don't need.

dmody, 2016-04-20:
Great advice everyone. Thanks so much!  

@duncan-a  You're probably right.  I could see myself in need of a prototyping board that you can plug the ESP-12 module into easily, do some programming and testing, then pull it out for final testing and use.  Does anyone know of such a beast?

duncan-a, 2016-04-20:
You can't "plug the ESP-12 module" easily (if you're thinking breadboards)
as they're 2mm spacing (as are most of the others).

You could use one of these (but fit 8-way stackable, long-pin, female
headers, not the male headers supplied - they span the whole space on a
breadboard).

http://www.banggood.com/ESP-07-ESP8266-Serial-WIFI-Module-with-IO-Adapter-Plate-p-980652.html

I have a couple of these which I prototype with and then design a PCB for a
soldered ESP-12 (or an ESP-07 on the rare occasions I want an external
antenna).

Remember that you'll need a USB-FTDI adaptor too.

dmody, 2016-04-20:
Well, I guess I see a nice business opportunity.  :)

duncan-a, 2016-04-20:
If you mean for breakout boards that are narrower, I've done that but never
had them manufactured - if you mean for a board using "Pogo Pins" or
something to hold an ESP-07/12/12E temporarily, I'm working on it...

I suspect the ESP-32 will probably take over quite a lot but, with two
processors, minimum current will be higher, plus there are all the new
additions which may take a toll.

philippe-gregoire, 2016-04-20:
re: 'fakir' board with pogo pins, there was such a thing featured on hackaday a while back.
I ordered the pogo pins (about 1$ for 50!) but didn't go as far as making anything than fiddle with them...

re: ESp-32, the beta boards are 1mm pitch, that's going to get beyond what regular DIYselfers like me can solder with their sturdy fingers...

Ah, also, there have been designs floating around where the ESP is powered through a switchable regulator (or maybe also a FET, if I recall well) controlled by a RTC chip. Typically, the ESP tells the RTC to switch its power off and re-wake up (power-on) later.

philippe-gregoire, 2016-04-20:
Re the pogo pins ESP board, see this: http://hackaday.com/2016/01/24/truly-versatile-esp8266-wifi-webcam-platform/

philippe-gregoire, 2016-04-20:
re: RTC clock controlling ESP's power: http://www.esp8266.com/viewtopic.php?f=13&t=5747

duncan-a, 2016-04-21:
That approach would work fine - if you need time-stamping or an interval
greater than can be obtained with an ESP. If you don't, then an RTC,
transistor and regulator might have a higher quiescent current than an ESP
in deep sleep (I don't know, just thinking out loud) - and why increase the
complexity if you don't need to?

duncan-a, 2016-04-21:
I'm thinking of setting the Pogo Pins in a block and having a clip-down
'lid' to retain the ESP, but basically the same...

I agree about 1mm pitch connections on the ESP-32 (bad eyesight and
unsteady hands are a chip killer - old age in my case), but there's bound
to be a 'mounted on a breakout board' version from China (maybe ExpressIf
will offer it as they already have the board).

jomego, 2016-05-03:
Someone has managed to advance ... I still consuming 7,8mA and I can not turn off the CP2102 ...

tinkermax, 2016-05-29:
This is what I did to reduce the current consumption of my v2 NodeMCU in deep sleep to < 50uA: https://tinker.yeoman.com.au/2016/05/29/running-nodemcu-on-a-battery-esp8266-low-power-consumption-revisited/

ginamdar, 2017-07-18:
[Hi please ignore if this sounds irrelevant/silly but does having Serial statements in the code causes uart chip to draw more current? I am asking this because I was testing with 3V and had same code with Serial logs and deepSleep enabled and wired. And it drained battery quickly. (Li-Ion 3.7 v 2500mAh) in less than 8 hours or so. 
Now same battery but with All Serial logs removed and its going since last night. Although i'm sure current draws in deepSleep in mA (have not measured yet) as proved above with AMS chip.

In addition im also measuring the voltage on A0 with oboard resistors 220K and 100K.

The code is simple (need to push to github) 
1) Read the D1 pin 
2) if high report the status over net
3) check battery status on A0 and report the value as is.
4) deepSleep(10*60*10e6) // 10 min.