esp8266/Arduino #720: Weird baudrate after deepSleep()

giannoug, 2015-08-25:
The following sketch produces weird output to the serial monitor, which is caused by the deepSleep function.

    void setup() {
      Serial.begin(115200);
    }

    void loop() {
      //Serial.begin(115200);
  
      Serial.println("Sup?");

      ESP.deepSleep(1 * 1000000);
      //delay(1000);
    }

giannoug, 2015-09-01:
Issue still persists. Device actually sleeps for the defined period, that's good. Tests were done on "hacked" ESP-01 devices, exactly like this photo: http://i.imgur.com/3CvqHkQ.jpg

Two of my ESPs are behaving the same way, a vanilla one doesn't work at all (as expected).

Any ideas? I can also see this behavior when not using Serial print statements.

igrr, 2015-09-01:
When ESP restarts, the bootloader prints some data at 74880 baud. If your
terminal is configured for 115200, you will see some garbage at startup.
This is normal behavior for the 8266 chip.

If you absolutely need to avoid garbage, you can use Serial.swap() function
(see reference docs for details).

giannoug, 2015-09-01:
Ah, this makes sense. Thank you!

closed 2015-09-01 by giannoug

drmpf, 2015-09-02:
how does Serial.swap() manage to work BEFORE the debug comes out?

igrr, 2015-09-02:
It doesn't — you simply connect your UART device/terminal to the swapped
pins. This way device/terminal only gets clean output.

drmpf, 2015-09-02:
That is difficult to do on an ESP8266-01

giannoug, 2015-09-02:
It is possible, I have managed to get 3 additional GPIOs on a ESP-01. You just need a steady hand, lots of patience, some flux and solder.