esp8266/Arduino #721: ESP.getVcc() doesn't read exact values

luciancerbu87, 2015-08-26:
So I have a wireless node sensor nodeMCU v1.0 with ESP12-E on it (http://www.aliexpress.com/item/2PCS-V3-NodeMcu-Lua-WIFI-development-board-based-ESP8266-LoLin-with-newest-firmware/32368521069.html).
The problem is with the received values from the ADC, for power supply voltage I get the following :
3.739
3.702
3.708
3.711
3.755
3.721
3.761
3.764
3.732
3.734
3.737
3.741
3.744
3.745
3.753
3.793
all these values were from the same source, 1minute delay between them with deep sleep mode in that minute.
It there any hardware thing I should do? to solder something on A0 pin?

Frogmore42, 2015-09-03:
I doubt the accuracy spec for the part (if it has one) is better than 1%.  Let's assume the "actual" voltage is 3.750.  The readings you are seeing are +0.043 to -0.048.  Let's say 0.05.  Now 0.05/3.75 is 1.33%.  Bottom line, you should only be looking at one digit past the radix (decimal) point, the rest is noise.  If you want, you can take several readings at the same time and average them and perhaps get a little more information/stable number.

pgollor, 2015-09-10:
I get with my modules (ESP8266-01 and ESP826612) the same trash. The voltage is 3.3V but the output from the `getVcc()` function is every time 3.7V. This is much more than 1% deviation (round about 10%). Maybe there is a systematic error?
Have anyone the same problem?

Greets
Kalle

luciancerbu87, 2015-09-10:
well,  after some very poor results I decided to supply my esp directly from a li-ion battery.  

for 4.05 volts on baterry it gives me from ADC value 3854

for me it works fine.

asetyde, 2015-09-10:
Also i use vcc and adc , but my mac arrive to max 1000ma on usb and is enough .
My advice is use a dc power supply when you use this type of sensor or many sensor or lcd .

pgollor, 2015-09-10:
I try different external power supplies and the best result is:

`getVcc()` | Multimeter | difference 
-------------|---------------|--------------
3.5 V | 3.3V | 6%

With 6% difference it is not usable.

wizard23, 2016-04-06:
I had problems getting reliable AD readings too. I had multiple devices available so I discovered that each esp gives different readings. Some ESPs are almost accurate but in general the value I was reading was always higher then when measured with a scope of multimeter. 

A good thing was that at least each individual ESP gives consistent reading. So my solution was to apply a voltage that is known (in my case 4.2V of a full lipo battery) to calibrate each esp. With calibration it was quite accurate.

lrmoreno007, 2016-04-06:
float voltaje=0.00f;  
voltaje = ESP.getVcc();
Serial.print(voltaje/1024.00f);

:+1:

minida28, 2017-02-25:
Hi,

I use NodeMcu v2, and I have set:

- [ ] wifi client activated (RF enabled?) 
- [ ] ADC_MODE(ADC_VCC);` added
- [ ] ADC pin of NodeMcu floating (nothing connected to it)

I got this with my nodemcu v2:
```
26.76 <--- Must be 33
26.76 <--- Must be 33
26.77 <--- Must be 33
26.76 <--- Must be 33
26.76 <--- Must be 33
```
Any thought?

I powered it from USB port, and measure it with my DMM, the voltage between 3.3v pin and GND is 3.29 volt.

jindrichsirucek, 2017-03-09:
I got the same values as minida28 for 3.3V
2716
2718
2718
2734
2718
2717
2718
2718
2736
2736
2736
2718

And values like this for 
2.8V
2240
2328
2236
2232
2235
2320

r1me, 2017-03-14:
Values reported by ESP are correct, you are probably not using a capacitor (close to VCC-GND). The voltage drop can be heavy without it. I just tried bare ESP-12E without capacitor: 3.3V -> 2.68V (similar values to yours). With 470uF -> 3.0V. 1000uF (or even more) will do the job.

jindrichsirucek, 2017-03-14:
Ok thanks for the tip, Ill check that out.. I have Lolin nodemcu V3 board.. I think there is some capacitor - but I will try it..
But I think there is actuallly no reason to have this serious voltage drop when esp is not using a wifi..

davidegironi, 2017-04-25:
The caps does not solve the problem in this case.
On nodemcu v3 and similar boards TOUT is connected to a 220k/100k divider, by datasheet vcc is only available when TOUT pin17 is suspended (floating), but that is not the case, on those board it is connected to GND by a 100k resistor.

jindrichsirucek, 2017-04-25:
Yes @davidegironi you are right - its not about capacitor its because of voltage divider at nodemcu board.. thx..

minida28, 2017-04-25:
So to solve the problem, we have to remove the voltage divider resistors, any one have tried it? :)

Also I just know that Wemos Mini R1 TOUT pin is not connected to voltage divider, according to [this schematic.](http://escapequotes.net/wp-content/uploads/2016/02/wemos-d1-mini-shematics.jpg)

davidegironi, 2017-04-25:
With the voltage divider removed from one of my nodemcu v3 board it's 3.07V, caps once again does not make any difference. It's not yet like the plain ESP12E, cause ADC does not work properly, it works like this issue here: https://github.com/esp8266/Arduino/issues/3168
I don't get why on that board ADC works like this.

minida28, 2017-04-25:
@davidegironi 
Thanks for your test result, but what voltage did you get before you remove the voltage divider?
I bet it should be lower with voltage divider in place, well,, at least according to jindrichsirucek and my test result above.

davidegironi, 2017-04-25:
@minida28
Almost 2.7V, depending on the board tested. The voltage divider is 100k to GND, 220k from the A0 nodumcu PIN.

flashko, 2017-07-29:
Hello. I own LoLin module. If A0 is not connected ESP.getVcc() returns 2720 (mV). Vcc is 3.290V measured with multimeter. Difference is probably from interference on input resistor divider 220k/100k with internal divider. This is eliminating if just connecting A0 with Vcc (i use green dupont wire for this test video) and then ESP.getVcc() returns 3290 - exact as measured. Please anyone test with different ESP8266 module and/or different Vcc. For me ESP.getVcc() works fine.
I use ESP8266 core v2.3.0 and Arduino IDE 1.6.11.

(I apologise for my english. I use Google translate.)

https://www.youtube.com/watch?v=diQNNNFjH1Q

fah, 2017-08-26:
Solution:
1. in you code do
  ```
    uint16_t v = ESP.getVcc();
    float_t v_cal = ((float)v/1024.0f);
    char v_str[10];
    dtostrf(v_cal, 5, 3, v_str);
    sprintf(v_str,"%s V", v_str);
    Serial.println(v_str);
```
2. connect A0 with VCC(3.3V)

Now you can read "3.3 V" on the console.

sanchosk, 2017-09-03:
Yesterday I discovered strange phenomenon. Vhen measuring Vcc after 20 seconds of operation, it measures exact values.
However, if I do the measurement few miliseconds after waking from deep sleep and connecting to wifi, I get erratic values with differences of -0.4V.
Looks like there is some slowly charging capacitor on the Vcc pin or something.
Needless to say, the Vcc is always 3V.

tablatronix, 2017-09-10:
> nor may we exceed 1.0V on the A0 pin!
The adc range wont measure above, but it might be protected against, worth checking into

fah, 2017-09-11:
@tablatronix ESP8266 is quite robust but a limit is a limit and it may damage the input in a way you don't see immediately. (Linearity,...)
@sanchosk Vcc does vary on startup depending on the other components you have attached. Measure with a DSO to see the truth. If you LDO is close to its limit you may have found the reason.

devyte, 2017-10-20:
I believe that the ADC input is limited to 0-1V range. To measure in the range 0-3.3V, board manufacturers put a resistor divider on the pin, to map the range. However, resistors have tolerance, the most common being 5%.
Use of the ADC input should always be calibrated for each specific device. It's very easy to do a linear calibration on the input value read.
In any case, there isn't much that can be done in this repo. Closing.

closed 2017-10-20 by devyte