esp8266/Arduino #722: [Solved] WiFi Client Example Issue

dexter0us, 2015-08-26:
Hello Everyone, 

I have been working on the example "WiFi Client" and when I uploaded it to ESP, the code was correct but the ESP was not getting correct responses, sometimes it gets a response and sometimes don't.
The main reason was the "delay(10)" 10ms delay @ line 74. while the ESP is in the delay, the response is received and when the ESP comes out of delay, it detects that there was no response.

So the best option I could find out to solve was to use "while(!client.available())" and the problem solved.
And obviously we can apply a millis() here to make it out of this while loop.

I hope it helps someone and please correct it in the example.

Regards,

Duality4Y, 2015-08-27:
maybe do:
```
while(!client.available()) delay(1);
```
to give the rtos some time while your waiting.

Frogmore42, 2015-09-03:
The reason the example code does not always work, is that it is waiting a fixed amount of time before checking if there is anything back.  If the server responds quickly this will work.  If the server has to think about it (or get some information from a database or some other server) the client can check for available before anything is available and then you will get nothing, because nothing is available.  The better check is:
1. The client is still connected
2. There is nothing available
3. A timeout has not expired
In this case delay, I also print a dot so I know it is delaying.  I chose 100ms for the delay and 5 or 10s for the timeout.  Looking at the logs, I often see two or three dots, which means it takes more 100-200ms for the server to respond much of the time.  Some people put in a fixed delay of 500ms before checking which is probably okay for most sites most of the time.

The tight loop without a delay could starve the WiFi code so at least follow what @Duality4Y said.

dexter0us, 2015-09-03:
Thankyou @Frogmore42  for the information, actually I am also printing dots but I've not applied the timeout check for myself, because sometimes my local server's responses gets slow. But I'll surely apply it soon.
Thankyou @Duality4Y .

closed 2015-11-05 by igrr

Count-The-Kans, 2016-10-19:
Hello dexter0us ,

    Can please tell about the code you were using previously. I am curious to know about the line @74 as I myself having the same issue with my code, but I cant figure out where to do the additions and editing. Also, Frogmore42 please tell me how to implement those checks you mentioned above. 

Thanks in advance

Frogmore42, 2016-10-20:
Sure, use something like this to connect:

            boolean connected;
            int counter = 0;
            do {
                connected = client.connect(server, 80);
                if (connected) { break; }
                client.stop();
                delay(1000);
                Serial.println("retry");
                counter++;
            } while (!connected && counter <5);

Send whatever the you need to send
Use this to wait till the response is available

                int c = '\0';
                unsigned long startTime = millis();
                unsigned long httpResponseTimeOut = 10000; // 10 sec
                while (client.connected() && ((millis() - startTime) < httpResponseTimeOut)) {
                    if (client.available()) {
                        c = client.read();
                        Serial.print((char)c);
                    }
                    else {
                        Serial.print(".");
                        delay(100);
                    }
                }
`

dexter0us, 2016-10-20:
Thanks @Frogmore42  for reply.
@Count-The-Kans : The method explained by @Frogmore42 is better than mine.