esp8266/Arduino #723: Extra slash in -I build option

DonKinzer, 2015-08-26:
The include option for the sdk/include has an extra slash between those two directory names.  Below is the compile command for my test sketch file (extra EOLs added for readability).  The double slash is in the third line.
C:\Users\Don\AppData\Roaming\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-g++
 -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__
 -IC:\Users\Don\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0/tools/sdk//include
 -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11
 -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_THING -DARDUINO_ARCH_ESP8266 -DESP8266
 -IC:\Users\Don\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\cores\esp8266
 -IC:\Users\Don\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\variants\thing
 C:\Users\Don\AppData\Local\Temp\build1303870529194311090.tmp\sketch_aug26b.cpp
 -o C:\Users\Don\AppData\Local\Temp\build1303870529194311090.tmp\sketch_aug26b.cpp.o

igrr, 2015-08-26:
Does the build fail on Windows? As far as i can tell, this is not an issue
on Linux and OS X.

DonKinzer, 2015-08-26:
Sorry, I should have indicated that the build completed without errors.  The double slash just caught my eye and I was surprised that it didn't cause a problem.  I'm running this on Windows 7 (native, i.e. not in a VirtualBox).

commit 60775dcd (2015-09-30): Remove extra slash in path

fix #723
closed 2015-09-30 by igrr