esp8266/Arduino #724: beginWPSConfig() is crashing my ESP8266 with call by reference

DaKaZ, 2015-08-26:
Hello,

First off - THANK YOU to all the contributors and maintainers here.  You adoption speed and support of the 1.3 SDK is truly impressive.

I am trying to get WPS Push Button support on my ESP8226 based IoT hardware.  When the button is pushed, I am calling `WiFi.beginWPConfig()` and then my board prints a few debug messages and then reboots.  So I added some more debugging messages and found it was crashing on the call back.

I looked at the WPS example in the 1.3 SDK and noticed that the SDK does not call by reference: `wifi_set_wps_cb(user_wps_status_cb);`

So I changed the callback in ESP8266WiFi.cpp to: `if(!wifi_set_wps_cb((wps_st_cb_t) wifi_wps_status_cb)) {` and now I get successful completion of WPS some of the time (I still get some crashes in the process).  My debug output looks like this:

```
WPS Button Pushed
Calling WiFi.beginWPSConfig()
state: 5 -> 0 (0)
rm 0
pm close 7 0 0/49634424
wifi evt: 1

stopAll 3ffe9edc 3ffe9edc
wps begin
wifi_wps_enable

****
calling wifi_wps_start
wps scan

****
calling esp yield

****
returned from esp yield
build public key start
build public key finish
f 0, scandone
wps cb status: 2

wps TIMEOUT

****
calling esp_schedule()
```
We need to figure out why this crashes 1/2 the time when when in `wifi_wps_start` even with the change to the call back to call by value not reference.  I am fairly new to Arduino development so I think the problem here might be that memory for the callback function is getting cleared since we have already left the `beginWPSConfig` method (it returns true long before the callback is called).

Next, assuming I actually connected to an access point successfully, how to do I retrieve the password so I can store it and use it the next time the system comes up?

Not sure what else I can do to get debug output to asset with the problem.  I am happy to work through this in realtime if someone has the expertise to guide me.  Thanks!

DaKaZ, 2015-08-27:
In further investigation, I think we have to pass the function call by reference, but this always crashes.  Here is the stack trace:
```
WPS Button Pushed
Calling WiFi.beginWPSConfig()
wps begin
wifi_wps_enable

****
calling wifi_wps_start
wps scan

****
calling esp yield
build public key start
build public key finish
beacon timeout
state: 5 -> 0 (1)
rm 0
wifi evt: 1

stopAll 3ffe9edc 3ffe9edc
Fatal exception (28):
epc1=0x40206fee, epc2=0x00000000
Exception (28):
epc1=0x40206fee epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000001c depc=0x00000000

ctx: sys 
sp: 3ffffd60 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff00:  3ffebbec 3fff12c4 3fffff20 40204cda  
3fffff10:  4023a4c8 00000028 3fff2058 4023a4d8  
3fffff20:  00000001 6f64754b 47326f73 3fff2c00  
3fffff30:  0000000a 4020b560 3ffeb2f0 0000000a  
3fffff40:  3fff2ce8 b6059408 c8fea607 4023c10f  
3fffff50:  3fff2070 3ffebaa0 68f642b6 60000600  
3fffff60:  3fff1790 3fff10e4 00000001 60000600  
3fffff70:  40214895 3ffebaf4 3ffebaa0 68f64186  
3fffff80:  402148da 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffebac8 3fffdab0 3ffeb38c 40202039  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
, epc3=0x00000000, excvaddr=0x0000001c, depc=0x00000000
b��d䄆����������ć䦦FC�����b愄������䄥Ĥ熤D�ƄbŧĄ�E���Ĥb�d
```

igrr, 2015-08-27:
Is there anything external to the 8266 necessary to test this WPS Push Button thing?
I.e. some wifi router which supports this mode?

DaKaZ, 2015-08-27:
Hi @igrr - of course to get through the entire process, yes, you need a router/access point with WPS Push Button support.  However, we should be able to complete the process (without connecting) with the unit crashing, it should just "fail" which is where I am at right now.

In particular, here is something else I noticed.  When the process DOES run through completion (which happens about 1 out of 10 times with the current code/pass by reference) the code does not run as I would expect.  What I mean is, it seems like calling `esp_yield()` should be a blocking event until the callback is called and `esp_schedule()` is called, then we return to the `beginWPSConfig()` method.  BUT - when we do run through completion, the flow looks like this:

1. call method `beginWPSConfig()`
1. call method `wifi_wps_enable()`
1. call method `wifi_wps_start()`
1. call method `esp_yield()`
1. *** we then immediately return and end the beginWPSConfig method, this should happen last
1. wps public keys are create
1. wps scan starts
1. wps TIMEOUT received (after 2 minutes)
1. call method `esp_schedule()`

igrr, 2015-08-27:
>    2. call method wifi_wps_enable()
>    3. call method wifi_wps_start()
>    4. call method esp_yield()
>    5. *** we then immediately return and end the beginWPSConfig method,
>    6. wps public keys are create
>    7. wps scan starts
>    8. wps TIMEOUT received (after 2 minutes)
>    9. call method esp_schedule()

Where are you calling beginWPSConfig from (setup/loop, pin interrupt, timer
interrupt)?

DaKaZ, 2015-08-27:
I call this in setup:
```
attachInterrupt(WPS_DIO, BUTTON_WPS_event, FALLING);
````
and here is my method for handling it:
```
    void BUTTON_WPS_event()
    {
      Serial.println("\nWPS Button Pushed");
      //BREAK_EVENT = 1;
      startFlashWPS(100, 120000); //2 minutes
      Serial.println("Calling WiFi.beginWPSConfig()");
      WiFi.mode(WIFI_STA); // WPS only works in station mode
      delay(1000);
      if ( WiFi.beginWPSConfig() )
      {
        Serial.println("WPS Success!");
      } else {
        Serial.println("WPS Failed");
        Server_init();
      }        
    }
```

igrr, 2015-08-27:
Okay, calling blocking functions from an interrupt is a bad idea.
Serial.println may cause a crash if called from interrupt, and
beginWPSConfig will certainly crash. Set a flag in the interrupt handler
and check the flag inside setup() or loop().

DaKaZ, 2015-08-27:
@igrr awesome.  I changed the code to set a global variable and then inside loop() based on that variable I kick off the WPS process.  I have run it about 10 times and its worked perfectly each time!  Sorry I am fairly new to embedded development so I did not realize that executing code after an interrupt was a back practice.

So, now to make it usable I have to complete the "todo" section to be able to return the WPS status to the main program AND somehow retreive the password (phasephase / wireless key).  Any ideas on that?  I also don't want to duplicate work so let me know if you've already started on the process.  Here was going to be my approach:

1. Add `_wpsStatus` to the ESP8266WiFiClass.
1. Create function:
   ```
   int  ESP8266WiFiClass::getWPSStatus(void) {
       return _wpsStatus;
   }
   ```
1. Store status at top of callback: `_wpsStatus = startus;`

Links2004, 2015-08-27:
as long you not call WiFi.begin at any point, the SDK keeps track of the SSID + PW no need to store it your self. the SSID and PW is stored in flash and survive even an update.

for SSID call:  ```WiFi.SSID();```
for the PW we not have a function, in theory you can get it form the SDK with ```wifi_station_get_config```
take a look here:
https://github.com/esp8266/Arduino/blob/d108a6ec30193a8e44eb9c72efc7a55853b54f09/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp#L356

note:
the ```wifi_wps_status_cb``` function don't have access to the class and is running in the "system" threat.
so adding a variable to the class will not work. if you use a global static in the file it shut work.

DaKaZ, 2015-08-28:
@Links2004 I don't seem to ever connect after a reset (example: power lost) unless I call WiFi.begin(...).  What am I am missing?

DaKaZ, 2015-08-28:
Since we have resolved the main crasher (ie: don't call Serial debugging during an interupt) I am now attempting to connect to an AP with WPS_PBC but the behavior is a bit odd.  The ESP8266 appears to connect as seen here:
```
WPS Button Pushed
Calling WiFi.beginWPSConfig()
bcn 0
del if1
mode : sta(18:fe:34:9f:e1:4f)
wps begin
wifi_wps_enable

****
calling wifi_wps_start
wps scan

****
calling esp yield
build public key start
build public key finish
f 0, wps discover [mywirelessssid]
scandone
WPS: neg start
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
process pubkey start
process pubkey finish
wps: ignore overlap identifier
wps: ignore overlap identifier
state: 5 -> 2 (2a0)
rm 0
wifi evt: 1

stopAll 3ffe9ec4 3ffe9ec4
```
The system just stops here although you'd think it had just configured itself.  Eventually, the WPS timeout kicks in:
```
wps cb status: 2

wps TIMEOUT

****
calling esp_schedule()

****
returned from esp yield

```
and we are not connected to the AP.  Any ideas / suggestions?

DaKaZ, 2015-08-28:
@igrr the WPS process is still a bit buggy.  Take a look at this stack trace after WPS negotiation (with sensitive info removed):
```
WPS Button Pushed
Calling WiFi.beginWPSConfig()
wps begin
wifi_wps_disable
wifi_wps_enable

****
calling wifi_wps_start
wps scan

****
calling esp yield
build public key start
build public key finish
f 0, wps discover [***removed****]
scandone
WPS: neg start
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
process pubkey start
process pubkey finish
WPS: key[***removed***]
wps finished
state: 5 -> 2 (2a0)
rm 0
wifi evt: 1

stopAll 3ffe9f2c 3ffe9f2c
Fatal exception (28):
epc1=0x4020700e, epc2=0x00000000, epc3=0x00
Exception (28):
epc1=0x4020700e epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000001c depc=0x00000000

ctx: sys 
sp: 3ffffcb0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe50:  3ffebc3c 00000002 3ffffe70 40204cfa  
3ffffe60:  4023a48c 3fff20a8 3fff20a8 4023a49c  
3ffffe70:  00000001 6f64754b 47326f73 00474747  
3ffffe80:  00000000 00000000 00000000 00000000  
3ffffe90:  00000000 4ce0000b 028f2020 4023abfe  
3ffffea0:  000000a0 3fff20a8 3ffed874 3ffed88e  
3ffffeb0:  3fff17e0 3fff1134 000002a0 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000000 00000000 00000000 00000000  
3ffffee0:  00000000 00000000 00000000 00000000  
3ffffef0:  00000000 00000000 00000000 00000000  
3fffff00:  00000000 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000027 00000000 3fff2818 4023a776  
3fffff30:  3ffefd80 3fff20a8 00000000 3fff17e0  
3fffff40:  3ffefd80 00000000 3ffed868 3ffed874  
3fffff50:  3ffed874 0000001e 00000000 00000027  
3fffff60:  00000000 402148b1 4021af7d 3ffefd80  
3fffff70:  3ffed868 000000f5 3ffec398 3ffec398  
3fffff80:  000000a0 3ffefd80 00000000 3fffdcb0  
3fffff90:  4021a928 3fffdab0 00000000 3fffdcc0  
3fffffa0:  3ffec398 40000f49 3fffdab0 40000f49  
<<<stack<<<
000000, excvaddr=0x0000001c, depc=0x00000000
b��d䄆����������ć䆦FC�b��d䄆����������ć䆦FC�����b
```

DaKaZ, 2015-08-28:
@Links2004 The good news is that even though the above crashed, after it connected with WPS it re-connected automatically just as you said when it rebooted!

DaKaZ, 2015-08-28:
@igrr @Links2004 Thank you both so much!  The above error and stack trace were not WPS related, it was my code.  I am now successfully connecting with wps_pbc and this is amazing!  Thank you again for all you do!!!

closed 2015-08-28 by DaKaZ