esp8266/Arduino #726: dtostrf adds extra space before decimal

lukecyca, 2015-08-27:
According to the Atmel reference manual, [dtostrf](http://www.atmel.com/webdoc/AVRLibcReferenceManual/group__avr__stdlib_1ga060c998e77fb5fc0d3168b3ce8771d42.html) has a width parameter defined as:
> minimum field width of the output string (including the '.' and the possible sign for negative values)
> width is signed value, negative for left adjustment".

The current implementation of this for esp8266 adds additional space in front of the decimal point when given a small (or negative) value for `width`. This is a regression — a previous version of this project did not have this issue. However I am not sure when it was introduced.

```c
f = -8.375;
dtostrf(f, 0, 3, tmp);  // "-8  .375"
dtostrf(f, -1, 3, tmp); // "-8   .375"
dtostrf(f, 1, 3, tmp);  // "-8 .375"
dtostrf(f, -2, 3, tmp); // "-8    .375"
dtostrf(f, 2, 3, tmp);  // "-8.375"
dtostrf(f, -7, 3, tmp); // "-8         .375"
dtostrf(f, 7, 3, tmp);  // " -8.375"

f = 3.625;
dtostrf(f, 0, 3, tmp);  // "3   .625"
dtostrf(f, -1, 3, tmp); // "3    .625"
dtostrf(f, 1, 3, tmp);  // "3  .625"
dtostrf(f, -2, 3, tmp); // "3     .625"
dtostrf(f, 2, 3, tmp);  // "3 .625"
dtostrf(f, -7, 3, tmp); // "3          .625"
dtostrf(f, 7, 3, tmp);  // "  3.625"
```

lukecyca, 2015-08-27:
Pinging @martinayotte, @Makuna, and @Links2004 who have worked on `dtostrf()` recently.

martinayotte, 2015-08-27:
Hi likecyca,
I've submitted a patch about 20 days ago related to dtostrf().
Are you actually using the HEAD of the github ? 
Because the previous version had so many bugs, as described in this PR, even a big one where sqrt(2) was shown as "1 .00000000002147483647" : 
https://github.com/esp8266/Arduino/pull/664
Maybe there are still some others bugs, since they so many use cases to check.
(BTW, this latest version was given to me by Odometer on Arduino forum, see also thread : http://forum.arduino.cc/index.php?topic=329949.45 )

lukecyca, 2015-08-28:
Thank you @martinayotte, it is indeed fixed with your patch. I had been using the stable/staging releases.

Great work on the improvement. Thanks again.

closed 2015-08-28 by lukecyca

paulhayes, 2015-11-19:
Would be great to see this roll out into stable. Had to switch my project to staging, yet it's been fixed for quite a while now.

martinayotte, 2015-11-19:
The stable version is outdated of July.
I think it is only that IGRR couldn't settled is mind about the stability of Staging and then to tag it as Stable.
There is so much new code in Staging needed for everyone that I consider it as the MainDev branch, while the Stable is some kind of "archive" branch.

paulhayes, 2015-11-19:
That's not good, especially when the Readme directs people to the "archive"
branch by default.

Also semantics are important.