esp8266/Arduino #728: True random number

Arilon, 2015-08-28:
Hi,

I just started to use this ESP8266 Arduino SDK. It is just awesome. Thanks for sharing it. 

In my project I need a true random number every time the node boots up. The problem is that the ESP8266 always starts the exact same state, resulting in the same random numbers even if you use randomSeed(SOMETHING) before.

The solution I like to share and discuss to this problem, is to initialize the randomSeed with the ESP8266 CPU cycle count since boot up. This is always different when it executes the random seed, resulting in a more or less true random number.

```
int getRandomNumber(int startNum, int endNum) {
  randomSeed(ESP.getCycleCount());
  return random(startNum, endNum);
}
```

What do you think?

duncan-a, 2015-08-28:
I've not yet needed a random number for anything but your novel solution
would seem to be the answer for the question I have yet to ask.

If it works, then it's problem solved...

mc-hamster, 2015-08-28:
For my cryptographic functions, I've been using the amount of time required to connect to an access point. That time is fairly random (within bounds) in itself. Using the CPU cycle count is another good method. Maybe I'll use that when the AP is unavailable.

- Jm

giannoug, 2015-08-28:
You can always use the ADC's value as a seed, its totally random everytime, provided that the input is floating.

mc-hamster, 2015-08-28:
Generally, it's best to avoid using that method of generating a seed. It's a dedicated hardware solution to something which, as illustrated above, has multiple solutions in software. That said, it's a good option to keep on your tool belt as a last resort in the case that the pure software solutions don't work.

- Jm

Links2004, 2015-12-30:
check: #1173
```cpp
/**
  Random Number Generator 32bit
  http://esp8266-re.foogod.com/wiki/Random_Number_Generator
 **/
 #define RANDOM_REG32  ESP8266_DREG(0x20E44)
```

closed 2015-12-30 by Links2004