esp8266/Arduino #731: softAP redirect all pages to 192.168.4.1 (DNS/redirect)

martin159, 2015-08-29:
I'm using ESP8266 as WiFi AP for initial client configuration.
Default IP of ESP8266 is 192.168.4.1, so client must open URL `http://192.168.4.1/` in their browser after connect to AP.

Is possible to set simple DNS server or any other redirect in ESP8266?
If user open url `http://google.com/` or `http://github.com/`.... ESP8266 will redirect/show own webserver at  `http://192.168.4.1/`.

Links2004, 2015-08-29:
what you searching is named Captive Portal
take a look here:
https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266/libraries/DNSServer/examples/

lrmoreno007, 2015-08-30:
You can try to set:

IPAddress ip(192, 168, 4, 1);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns(192, 168, 1, 1);

After this in setup void:

WiFi.config(ip, gateway, subnet, dns);

I think this should work too.

closed 2015-09-08 by igrr