esp8266/Arduino #732: OSX - Nodemcu v1.0 not showing on Arduino IDE port menu

cescoferraro, 2015-08-30:
Just downloaded the latest Arduino IDE 1.6.5
Added the boards manager [tried with both staging and stable]
Downloaded SiLabsUSBDriverDisk and ch341ser_mac drivers.
Rebooted. I can see all the esp8266 types under boards menu.
But under ports i cant see my board. What am I doing wrong??

asetyde, 2015-08-30:
Check if Mac see com port , sometimes Mac put default Bluetooth serial port

cescoferraro, 2015-08-30:
Tried to upload sketch to the bluetooth incoming port but got.
warning: espcomm_sync failed
error: espcomm_open failed

Osx does not seem to recognize the module.
mac-osx :: ~ » ls -l /dev/cu*
crw-rw-rw-  1 root  wheel   18,   1 Aug 30 09:54 /dev/cu.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   18,   3 Aug 30 09:54 /dev/cu.Bluetooth-Modem

asetyde, 2015-08-30:
No sorry Francesco , you must use usb serial ,

If my cable on Mac it's not link , you see this

cescoferraro, 2015-08-30:
See what? 
I should see an extra serial port at the menu right by using the android cable, right?
I dont need any serial cables since the board has its own FTDI, right?

cescoferraro, 2015-08-30:
YEAH! tried SEVEN cables. With the former, now I see SLAB_USBto UART on the port menu!
But still cant program it. I think because I cant see esptool on the programmer menu.

I get:
Uploading 205232 bytes from /var/folders/lg/jnvy9my56nsgqkkxt3ftqwz40000gn/T/build1264847574822280989.tmp/Blink.cpp.bin to flash at 0x00000000
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: didn't receive command response

asetyde, 2015-08-30:
Wait nodemcu has not ftdi chip

igrr, 2015-08-30:
esptool should not appear in programmers menu. Apparently some outdated
tutorials still mention it.

asetyde, 2015-08-30:
You must check your driver chip .
I've also mac but i prefer Windows 

Inviato  dal mio iPhone

cescoferraro, 2015-08-30:
I have enabled verbose. It looks like the esptool is missing something.
I have never used the nodemcu v1.0 before. should I press something before flashing?
Something happened after trying to upload , since the LED is not blinking anymore. I think I have wipped out LUA, but did not quite get my arduino blink sketch running

esptool v0.4.5 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 9600
	setting port from /dev/tty.usbserial to /dev/cu.SLAB_USBtoUART
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/lg/jnvy9my56nsgqkkxt3ftqwz40000gn/T/build1264847574822280989.tmp/Blink.cpp.bin success
opening port /dev/cu.SLAB_USBtoUART at 9600
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_open
Uploading 205232 bytes from /var/folders/lg/jnvy9my56nsgqkkxt3ftqwz40000gn/T/build1264847574822280989.tmp/Blink.cpp.bin to flash at 0x00000000
	erasing flash
	size: 0321b0 address: 000000
	first_sector_index: 0
	total_sector_count: 51
	head_sector_count: 16
	adjusted_sector_count: 35
	adjusted_size: 023000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 10000
	setting character timeout 100
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm_send_command: receiving 2 bytes of data
	writing flash
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: didn't receive command response
closing bootloader

igrr, 2015-08-30:
Normally the board should be reset automatically during upload. I'm using
nodemcu 1.0 on a Mac, and it works quite well... I must say, I had issues
like this with some usb cables as well.
If this issue is 100% reproducible with other cables as well, we can try to
increase flash erase timeout.

asetyde, 2015-08-30:
With my mac all ok

cescoferraro, 2015-08-30:
Got it now! I got confused because on the chip its written to use 9600 baud rate. [I think its LUA related]
Changed do 115200 and it got it to upload. 
I have also deleted the stable and downloaded the staging version at the same time.

asetyde, 2015-08-30:
✌️

mtnbrit, 2015-08-30:
I love being able to upload at 921600 baud with this board. 

One thing I noticed with the v1.0 board is that ESP.reset() doesn't work, it just hangs up showing wdt but never restarts. Perhaps they are pulling rst low to hard or something? ESP.reset() works fine on the v0.9 board for example.

asetyde, 2015-08-30:
Esp.restarts work but not very well , I say to you my method , 

- board never reset when you upload code , you need after upload , disconnect and give power to enter in normal situation. 

- my function put all pin in standard  start status 
All pin input and hight 
- never use or be safe on gpio 0 and 2 , be safe not use

Links2004, 2015-08-30:
@mtnbrit
```ESP.reset();``` does kill the SDK hard but not do a power loop ( resenting registers ) this can end in bad states. like your wdt loop.
for safe reboot use ```ESP.restarts();```

see here:
https://github.com/esp8266/Arduino/blob/260ab630f3391ee8bd6c6fed7636bad15a819673/hardware/esp8266com/esp8266/cores/esp8266/Esp.cpp#L116

mtnbrit, 2015-08-30:
I tried both ESP.restart and ESP.reset but the both ended in wdt but no actual restart on NodeMCU v1.0 board

asetyde, 2015-08-30:
Please check gpio 0 , 2   Must hight
My Board reboot , not agenzia you flash board

closed 2015-08-31 by igrr

francwalter, 2017-06-26:
Hello
sorry when I put my answer here, maybe  wrong:
I have the same problem with a [NodeMCU V3 ESP8266 ESP-12 E Lua CH340](http://www.ebay.de/itm/162472635911) - when I put in the USB-cable nothing appears in the port list, but when I put there the good old Arduino Uno V3 my port appears and all works. I tried it with 1.6.5 and the actual 1.8.3. The port submenu is grey in 1.6.5 and shows only the usb bluetooth in 1.8.3.
I tried two different USB cables which are good working but no difference.
The ESP8266 shortly blinks blue when I put the cable to my MacBook (with OS 10.11.6).
I installed the ESP8266 IDE extension for the Arduino IDE.
What can it be?
Thanks.
frank

mtnbrit, 2017-06-26:
You need to install the CH340 driver for OS X.

francwalter, 2017-06-26:
Thank a lot!
Works now 👍 with the driver [CH341SER_MAC-1.4](http://blog.sengotta.net/wp-content/uploads/2015/11/CH341SER_MAC-1.4.zip) from the site "[Signed Mac OS Driver for Winchiphead CH340 serial bridge](https://blog.sengotta.net/signed-mac-os-driver-for-winchiphead-ch340-serial-bridge/)"
I also had to set in Arduino IDE the Reset Method to nodemcu (in Tools) instead of ck and then it worked.
Shows now in the statusbar: 
**Generic ESP8266 Module, 80 MHz, 40 MHz, DIO, 115200, 512K (64K SPIFFS), nodemcu, Disabled, Node on /dev/cu.wchusbserial620**
and I can upload sketches :)

rafaeleyng, 2017-10-12:
@francwalter Hello there! I'm having the same issues, and your solution, unfortunately, didn't work for me.

I'm on macOS Sierra 10.12.6

I'm trying to setup a NodeMCU 1.0 board, and the port don't appear on the list, even after installing CH341SER_MAC-1.4.

The error I get is:

```
/Users/rafael/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9/esptool -vv -cd nodemcu -cb 115200 -cp /dev/cu.usbmodem1421 -ca 0x00000 -cf /var/folders/m3/9tc0dcj531d1v746xyl_ms7h0000gn/T/arduino_build_504741/NodeMCU-Blink.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.usbmodem1421
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.usbmodem1421 at 115200
error: cannot access /dev/cu.usbmodem1421

error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
```

The README of this driver states:

> After the installation is completed, you will find serial device in the device list(/dev/tty.wchusbserial*), and you can access it with serial tools.

but this file doesn't exist there.

Also, about your phrase:

> I also had to set in Arduino IDE the Reset Method to nodemcu (in Tools) instead of ck and then it worked.

I couldn't find anything related to that in Arduino IDE. Could you provide some screenshot, maybe?

---

PS: tried on Windows and, while it doesn't give an error (it actually does the upload up to 100%), I'm unable to see any result. Like, running a simple led blinking doesn't make the led actually blink.

francwalter, 2017-10-13:
What exactly for Screenshot do you need?

rafaeleyng, 2017-10-16:
I wanted to see where was the option to set:

> I also had to set in Arduino IDE the Reset Method to nodemcu (in Tools) instead of ck and then it worked.

But I ended up using a Windows machine and everything worked, so I gave up trying to make it work on mac.

francwalter, 2017-10-16:
[image]
This is it. But I think that is my fault, this "Reset" menu-entry is only shown when set to "Generic ESP8266 Module" and not to  "NodeMCU 1.0 ..." where I guess this setting is implicit.

But anyway, if the driver is not installed and not visible under Port after plugged into the USB-Jack:
[image]
it won't work. 
By the way, each USB-Jack has his own port-name on my MacBook with OS 10.11.6

rdedinsky, 2018-09-03:
This driver worked for me:
https://kig.re/2014/12/31/how-to-use-arduino-nano-mini-pro-with-CH340G-on-mac-osx-yosemite.html

FunMiles, 2018-09-14:
I have a similar situation but a bit more twisted.
I installed the Silab drivers and I was able to plug in my NodeMCU and program it. I did run the blink program in two ways. One with the on-board LED and one with an outside LED connected to a pin and ground.
Then I decided to try a WiFi http program. It seemed to work, but when I opened the serial console, the output was garbled as the console was running 9600 bauds instead of the 115k set for the program upload. I switched it to 115k and decided to unplug then replug the node mcu.
Since then, the port never shows up anymore. I've tried swapping three other micro-USB cables and nothing works so far.
I've un-installed and re-installed the Silab drivers twice as well.
Any way to see outputs from the drivers that would indicate whether it is invoked or rejects the board?

thelinguist, 2019-03-07:
I had the same problem as the original poster. When you are on a mac, and the board isn't appearing in the port, you may have forgotten to give permission for the driver to run. I was using Silab's driver, and although I gave permission for the INSTALLER to run, I did not give permission for the DRIVER to run (annoying!) Mac will block the driver from running after you install it (I was on OSX 10.13.6). If you have this problem, make sure you grant permission for the driver to run in System Preferences:
[image]
I wasted several hours wondering why my board wasn't showing up in the ports menu, I hope this helps someone

ghabreu, 2019-03-15:
> I had the same problem as the original poster. When you are on a mac, and the board isn't appearing in the port, you may have forgotten to give permission for the driver to run. I was using Silab's [...]

Thanks, that was it. If you're using MacOS Mojave, you need to give permission for the driver to run.

rafaeleyng, 2019-06-03:
This:

[image]

doesn't show up for me on Mojave. Do you know how do I force it to appear? @thelinguist @ghabreu

ghabreu, 2019-06-04:
> This:
> 
> [image]
> 
> doesn't show up for me on Mojave. Do you know how do I force it to appear? @thelinguist @ghabreu

It should appear right after installing / using the driver (causing it to be blocked). You may have to unlock the panel before under System Preferences > Security & Privacy.

[image]

nileshku, 2020-07-26:
Thread really helped. 
I am using board with cp2102 
Link to driver download [Link](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers)